Xilinx/qemu #6: Where is xilinx-master branch ?

ynsta, 2015-04-16:
Hello, the xilinx-master branch disappeared, it was the only version of qemu I managed to make work for zynq with device tree.

Furthermore xilinx wiki http://www.wiki.xilinx.com/qemu reference this branch. Could you please provide back this branch or update the wiki to point to a working version of qemu.

Regards

hpeyerl, 2015-07-08:
I see there's been no response to this since April.  I also can't get this to work.  The only branch I can see is pub/2014.2.plnx and it segfaults when trying to boot the kernel.

[8360624.993430] qemu-system-arm[9896]: segfault at 20 ip 00007fc0851bea24 sp 00007fc07f073600 error 4 in qemu-system-arm[7fc084fd7000+437000]
[8360634.689466] qemu-system-arm[9901]: segfault at 20 ip 00007f6e133afa24 sp 00007f6e0d264600 error 4 in qemu-system-arm[7f6e131c8000+437000]
[8360728.685912] qemu-system-arm[9907]: segfault at 20 ip 00007f0a5bf5fa24 sp 00007f0a55e14600 error 4 in qemu-system-arm[7f0a5bd78000+437000]
[8360785.428107] qemu-system-arm[9910]: segfault at 20 ip 00007f58fceeaa24 sp 00007f58f6d9f600 error 4 in qemu-system-arm[7f58fcd03000+437000]
[8361391.901214] qemu-system-arm[9923]: segfault at 20 ip 00007f60b9cf1a24 sp 00007f60b3ba6600 error 4 in qemu-system-arm[7f60b9b0a000+437000]
[8361400.563271] qemu-system-arm[9928]: segfault at 20 ip 00007feb2e9fba24 sp 00007feb288b0600 error 4 in qemu-system-arm[7feb2e814000+437000]
[8445967.420991] qemu-system-arm[31435]: segfault at 20 ip 00007f9479d86a24 sp 00007f9473c3b600 error 4 in qemu-system-arm[7f9479b9f000+437000]

alistair23, 2015-07-09:
Hello Stany and hpeyerl,

Sorry about the confusion between the wiki and the Git branches. We are working on this and will have an update for both in the next week. Hopefully that will fix your seg fault issue. If it doesn't let me know and we will see what we can do.

alistair23, 2015-07-13:
I just pushed an update to QEMU and the wiki has been updated to shown the new instructions. I'm going to close this issue now, if you have any problems feel free to open up a new issue.

closed 2015-07-13 by alistair23

hpeyerl, 2015-07-14:
Thanks for the update.  I've switched to the 2015.2.plnx branch and built it.  I encountered a lot of (-Werr) warnings and hacked them.. I'm not submitting a pullup req because many of them are probably not what you want.  I can submit a pullup if you like but here's the diff in the meantime:

diff --git a/hw/arm/arm_generic_fdt.c b/hw/arm/arm_generic_fdt.c
index dcb7d33..fcaef30 100644
--- a/hw/arm/arm_generic_fdt.c
+++ b/hw/arm/arm_generic_fdt.c
@@ -137,8 +137,8 @@ static void arm_generic_fdt_init(MachineState *machine)
 {
     ram_addr_t ram_kernel_base = 0, ram_kernel_size = 0;
 
-    void *fdt;
-    void *sw_fdt;
+    void *fdt = NULL;
+    void *sw_fdt = NULL;
     int fdt_size, sw_fdt_size;
     const char *dtb_arg, *hw_dtb_arg;
     char node_path[DT_PATH_LENGTH];
diff --git a/hw/arm/boot.c b/hw/arm/boot.c
index c7f8d0e..8d750f4 100644
--- a/hw/arm/boot.c
+++ b/hw/arm/boot.c
@@ -614,7 +614,7 @@ void arm_load_kernel(ARMCPU *cpu, struct arm_boot_info *info)
     int kernel_size;
     int initrd_size;
     int is_linux = 0;
-    uint64_t elf_entry, elf_low_addr, elf_high_addr;
+    uint64_t elf_entry;
     int elf_machine;
     hwaddr entry, kernel_load_offset;
     int big_endian;
diff --git a/hw/block/m24cxx.c b/hw/block/m24cxx.c
index a965c00..5d104ca 100644
--- a/hw/block/m24cxx.c
+++ b/hw/block/m24cxx.c
@@ -21,6 +21,7 @@
 #include "hw/i2c/i2c.h"
 #include "hw/hw.h"
 #include "sysemu/blockdev.h"
+#include "sysemu/block-backend.h"
 
 #ifndef M24CXX_DEBUG
 #define M24CXX_DEBUG 0
diff --git a/hw/core/fdt_generic.c b/hw/core/fdt_generic.c
index b6d21d1..514c4a2 100644
--- a/hw/core/fdt_generic.c
+++ b/hw/core/fdt_generic.c
@@ -31,6 +31,7 @@
 #ifndef FDT_GENERIC_ERR_DEBUG
 #define FDT_GENERIC_ERR_DEBUG 0
 #endif
+#define LOG_LEVEL_MASK(x) (1<<x)       // Where is this normally defined?
 #define DB_PRINT(...) do { \
     if (FDT_GENERIC_ERR_DEBUG) { \
         qemu_log_mask(LOG_FDT | LOG_LEVEL_MASK(2), ": %s: ", __func__); \
diff --git a/hw/core/fdt_generic_util.c b/hw/core/fdt_generic_util.c
index cc2c703..b9034da 100644
--- a/hw/core/fdt_generic_util.c
+++ b/hw/core/fdt_generic_util.c
@@ -628,7 +628,8 @@ static void trim_xilinx_version(char *x)
         }
         if (DIGIT(x[1])) {
             char *p;
-            strtol(x+1, &p, 0);
+            if (strtol(x+1, &p, 0))
+               p = p;
             if ( *p == '.') {
                 *x = 0;
                 return;
diff --git a/hw/misc/arm-smmu.c b/hw/misc/arm-smmu.c
index e1b2997..6b73166 100644
--- a/hw/misc/arm-smmu.c
+++ b/hw/misc/arm-smmu.c
@@ -6145,7 +6145,7 @@ static void smmu_ptw64(SMMU *s, unsigned int cb, TransReq *req)
     unsigned int outputsize;
     unsigned int grainsize = -1;
     unsigned int stride;
-    unsigned int level;
+    unsigned int level = 0;
     unsigned int firstblocklevel = 0;
     unsigned int tg;
     unsigned int ps;
diff --git a/hw/misc/blob-loader.c b/hw/misc/blob-loader.c
index a4b0137..608c76b 100644
--- a/hw/misc/blob-loader.c
+++ b/hw/misc/blob-loader.c
@@ -43,7 +43,7 @@ extern AddressSpace *loader_as;
 static void blob_loader_realize(DeviceState *dev, Error **errp)
 {
     BlobLoaderState *s = BLOB_LOADER(dev);
-    int size;
+    int size = 0;
     hwaddr entry;
 
     if (s->cpu_nr != BLOB_LOADER_CPU_NONE) {
diff --git a/hw/sd/ronaldo-sdhci.c b/hw/sd/ronaldo-sdhci.c
index 423f3f5..a262d39 100644
--- a/hw/sd/ronaldo-sdhci.c
+++ b/hw/sd/ronaldo-sdhci.c
@@ -82,7 +82,6 @@ static void ronaldo_sdhci_realize(DeviceState *dev, Error **errp)
 {
     DeviceClass *dc_parent = DEVICE_CLASS(RONALDO_SDHCI_PARENT_CLASS);
     RonaldoSDHCIState *s = RONALDO_SDHCI(dev);
-    SDHCIState *ss = SYSBUS_SDHCI(dev);
     DriveInfo *di_sd;
     DriveInfo *di_mmc;

alistair23, 2015-07-14:
Thanks hpeyerl,
None of those issues should cause any build or run problems (just some GCC warnings), so they aren't critical. In saying that no-one likes warning and if you submit a pull request I'll be happy to accept it and update the code.

alistair23, 2015-07-14:
Just make sure that it complies with the QEMU style. You have an if case in there without any {} following it, which is against the style.

hpeyerl, 2015-07-14:
I wasn't fixing for cleanliness.  But configure has:

configure:    QEMU_CFLAGS="-Werror $QEMU_CFLAGS"

so the above caused the compile to fail.

Sadly, my segfault hasn't gone away.  I can get qemu to boot zynq-14.5 but not my own kernel from xilinx git repo, and a 706 DTB.  Something in the DTB is causing it to fail.  If I boot the zynq-14.5 kernel with a 706 DTB that I compiled with dtc, it fails the same way.  There's also no console output so I have to dump __log_buf to see what it's doing.

I'm new to Zynq; have worked lots with OMAP.

alistair23, 2015-07-14:
Good point! I always compile with --disable-werror, which is probably a bad idea. If you send a pull request then I will definitely commit it.

Ok, if you can send me the images and command line arguments that you are using I can investigate and fix the issue for you. You can email them to me at: alistair.francis@xilinx.com if you don't want to put them on GitHub