Xilinx/qemu #10: what's

moypray, 2015-12-03:


alistair23, 2015-12-10:
Hello moypray,

There are quite a few differences between the Xilinx QEMU and the mainline QEMU. Xilinx recommends that you use the Xilinx QEMU as it has better support for our products. Some of the key features that the Xilinx QEMU repo has compared to the mainline repo are:
 - FDT Generic (Automatically generate machines from the device tree)
 - Improved MicroBlaze, Zynq and ZynqMP support
 - Support for more Xilinx peripherals
    - Including Display Port and the PMU
 - Improved ARMv8 EL support
 - Device loading improvements targeted for ZynqMP

Thanks,

Alistair

closed 2015-12-10 by alistair23

moypray, 2015-12-11:
@alistair23 , Got it, thanks for your information~~