Xilinx/qemu #12: Xilinx-Zynq does not compile in master branch

Afra-Felix, 2016-01-15:
I am trying to get the SMP support for qemu-system-arm for machine "xilinx-zynq-a9". I see that the main branch does not compile the file even though the file hw/arm/xilinx_zynq.c  there in the folder.
In the big_endian branch the patch for SMP is not there but it compiles with  "xilinx-zynq-a9" support.

alistair23, 2016-01-15:
Hello Krishna,
Thank you for bringing this to my attention. Somewhere along the way the Zynq device was removed from the Makefile, which is why you aren't seeing it. I am working on returning the Zynq machine to Xilinx's QEMU. Unfortunately there are a few other errors there that I am fixing. Hopefully I should be able to push the patch for this early next week.

In the meantime, I would recommend running Zynq using our automatic machine generation, you can find more details about it here: http://www.wiki.xilinx.com/QEMU+-+Zynq-7000

I'm sorry for the problems this has caused, thanks again for bringing it up.

alistair23, 2016-01-22:
Thank you for bringing that to my attention. I have pushed some patches which restore the hardcored Zynq-7000 machine.

Afra-Felix, 2016-02-04:
Hi Alistair,

There seems to be a bug which makes the qemu crash instead of booting up. So instead of creating a new issue I will continue from this .

I compiled the latest tree and I am getting the below error. Can you kindly look in to that?

qemu-system-arm -machine xilinx-zynq-a9  -m 1024 -kernel ./fsimage/u-boot  -serial null -serial mon:stdio -sd ./fsimage/sd.img   -nographic 
--------------------------------------------------------------------
Xilinx QEMU Feb  4 2016 11:31:39.
--------------------------------------------------------------------
qemu: hardware error: arm_mptimer: num-cpu 1 but this cpu is 1!

CPU #0:
R00=00000001 R01=00000000 R02=00000020 R03=e0001000
R04=00000000 R05=0405955c R06=04059530 R07=04059558
R08=00000000 R09=013ffe88 R10=00000000 R11=00000000
R12=e0001000 R13=013ffe60 R14=04025cb0 R15=04025bc8
PSR=200001d3 --C- A svc32
s00=00000000 s01=00000000 d00=0000000000000000
s02=00000000 s03=00000000 d01=0000000000000000
s04=00000000 s05=00000000 d02=0000000000000000
s06=00000000 s07=00000000 d03=0000000000000000
s08=00000000 s09=00000000 d04=0000000000000000
s10=00000000 s11=00000000 d05=0000000000000000
s12=00000000 s13=00000000 d06=0000000000000000
s14=00000000 s15=00000000 d07=0000000000000000
s16=00000000 s17=00000000 d08=0000000000000000
s18=00000000 s19=00000000 d09=0000000000000000
s20=00000000 s21=00000000 d10=0000000000000000
s22=00000000 s23=00000000 d11=0000000000000000
s24=00000000 s25=00000000 d12=0000000000000000
s26=00000000 s27=00000000 d13=0000000000000000
s28=00000000 s29=00000000 d14=0000000000000000
s30=00000000 s31=00000000 d15=0000000000000000
s32=00000000 s33=00000000 d16=0000000000000000
s34=00000000 s35=00000000 d17=0000000000000000
s36=00000000 s37=00000000 d18=0000000000000000
s38=00000000 s39=00000000 d19=0000000000000000
s40=00000000 s41=00000000 d20=0000000000000000
s42=00000000 s43=00000000 d21=0000000000000000
s44=00000000 s45=00000000 d22=0000000000000000
s46=00000000 s47=00000000 d23=0000000000000000
s48=00000000 s49=00000000 d24=0000000000000000
s50=00000000 s51=00000000 d25=0000000000000000
s52=00000000 s53=00000000 d26=0000000000000000
s54=00000000 s55=00000000 d27=0000000000000000
s56=00000000 s57=00000000 d28=0000000000000000
s58=00000000 s59=00000000 d29=0000000000000000
s60=00000000 s61=00000000 d30=0000000000000000
s62=00000000 s63=00000000 d31=0000000000000000
FPSCR: 00000000
CPU #1:
R00=00000000 R01=0000767b R02=f8f00000 R03=ffffffff
R04=00000000 R05=04059550 R06=04059530 R07=0405954c
R08=00000000 R09=013ffe88 R10=00000000 R11=00000000
R12=00000000 R13=013ffe60 R14=04039da0 R15=04001ae0
PSR=600001d3 -ZC- A svc32
s00=00000000 s01=00000000 d00=0000000000000000
s02=00000000 s03=00000000 d01=0000000000000000
s04=00000000 s05=00000000 d02=0000000000000000
s06=00000000 s07=00000000 d03=0000000000000000
s08=00000000 s09=00000000 d04=0000000000000000
s10=00000000 s11=00000000 d05=0000000000000000
s12=00000000 s13=00000000 d06=0000000000000000
s14=00000000 s15=00000000 d07=0000000000000000
s16=00000000 s17=00000000 d08=0000000000000000
s18=00000000 s19=00000000 d09=0000000000000000
s20=00000000 s21=00000000 d10=0000000000000000
s22=00000000 s23=00000000 d11=0000000000000000
s24=00000000 s25=00000000 d12=0000000000000000
s26=00000000 s27=00000000 d13=0000000000000000
s28=00000000 s29=00000000 d14=0000000000000000
s30=00000000 s31=00000000 d15=0000000000000000
s32=00000000 s33=00000000 d16=0000000000000000
s34=00000000 s35=00000000 d17=0000000000000000
s36=00000000 s37=00000000 d18=0000000000000000
s38=00000000 s39=00000000 d19=0000000000000000
s40=00000000 s41=00000000 d20=0000000000000000
s42=00000000 s43=00000000 d21=0000000000000000
s44=00000000 s45=00000000 d22=0000000000000000
s46=00000000 s47=00000000 d23=0000000000000000
s48=00000000 s49=00000000 d24=0000000000000000
s50=00000000 s51=00000000 d25=0000000000000000
s52=00000000 s53=00000000 d26=0000000000000000
s54=00000000 s55=00000000 d27=0000000000000000
s56=00000000 s57=00000000 d28=0000000000000000
s58=00000000 s59=00000000 d29=0000000000000000
s60=00000000 s61=00000000 d30=0000000000000000
s62=00000000 s63=00000000 d31=0000000000000000
FPSCR: 00000000
Aborted

alistair23, 2016-02-05:
Thanks for pointing this out. I am re-opening the issue. I have found and fixed the problem internally. Depending on reviews I will have a fix to you tomorrow or next week. Do you need this urgently?

Afra-Felix, 2016-02-05:
Thanks for update, Next week is fine with me

alistair23, 2016-02-12:
Sorry about the delay there. It is fixed now. Thanks for pointing this out

closed 2016-02-12 by alistair23

Afra-Felix, 2016-02-15:
Hi Alistair,

I was able to compile but when I try to run qemu im getting this error. Now It may be due to some library version issue in my Linux environment.Let me know if I need to make any change

$ ./arm-softmmu/qemu-system-arm 
--------------------------------------------------------------------
Xilinx QEMU Feb 15 2016 10:29:45.
--------------------------------------------------------------------
**
ERROR:/home/user/work/fixdqemu/qemu/qom/object.c:165:type_get_parent: assertion failed: (type->parent_type != NULL)

alistair23, 2016-02-16:
Yep, I see the problem. Pushing a fix now.

I would recommend that you use the aarch64-softmmu target instead of the arm-softmmu, as that is what we specifically support. You can run still run Zynq (ARMv7) machine on the aarch64-softmmu target.

Afra-Felix, 2016-02-17:
Thanks for update. The reason for trying arm-softmmu is that your repo has a patch for trying smp with 2 core and I wanted to get that feature in working.

Afra-Felix, 2016-02-17:
Hi I tried the fix. Uboot boots properly but it does not detect the MMC device when I pass using -sd .
Below is the log
--------------------------------------------------------------------
Xilinx QEMU Feb 17 2016 10:09:59.
--------------------------------------------------------------------
Warning: nic cadence_gem.1 has no peer

U-Boot 2015.10 (Feb 10 2016 - 15:24:43 +0100)

Model: Zynq Zed Development Board
Board:	Xilinx Zynq
I2C:   ready
DRAM:  ECC disabled 512 MiB
MMC:   zynq_sdhci: 0
Using default environment

In:    serial
Out:   serial
Err:   serial
Model: Zynq Zed Development Board
Board:	Xilinx Zynq
Net:   Gem.e000b000
Warning: Gem.e000b000 (eth0) using random MAC address - 52:50:93:93:ae:f7

Hit any key to stop autoboot:  0 
Card did not respond to voltage select!
** Bad device mmc 0 **
## Starting application at 0x00100000 ...
undefined instruction
pc : [<013ffbd4>]	   lr : [<1ff4d65c>]
reloc pc : [<e54b7bd4>]	   lr : [<0400565c>]
sp : 1eb24fa0  ip : 00000030	 fp : 00000000
r10: 00000002  r9 : 1eb27ef8	 r8 : 1ffa4ce0
r7 : 1eb35f88  r6 : 00100000	 r5 : 00000002  r4 : 1eb35f8c
r3 : 00100000  r2 : 1eb35f8c	 r1 : 1eb35f8c  r0 : 00000001
Flags: nZCv  IRQs off  FIQs off  Mode SVC_32
Resetting CPU ...

resetting ...

U-Boot 2015.10 (Feb 10 2016 - 15:24:43 +0100)

alistair23, 2016-02-17:
Can you include all of your command line and the images you are using?

Afra-Felix, 2016-02-19:
This is the command line for Qemu

/opt/qemu-2.5.0/bin/qemu-system-arm -machine xilinx-zynq-a9 -m 1024  -kernel /home/user/work/fsimage/u-boot  -sd /home/user/work/fsimage/sd.img -serial null -serial mon:stdio -nographic

The sd.img is a 64MB Fat formatted image which contains a QNX image which comes part of Xilinx BSP. This already works fine with Qemu 2.5.0.

alistair23, 2016-02-19:
Which BSP? Can you email me a link to the image? It is hard to debug without a reproducible test case. alistai <at> xilinx <dot> com

Afra-Felix, 2016-02-22:
The problem is independant of the BSP. All you need is just compile the u-boot 
wget ftp://ftp.denx.de/pub/u-boot/u-boot-2015.10.tar.bz2
tar xvf u-boot-2015.10.tar.bz2
cd u-boot-2015.10
Set the CROSS_COMPILE with ARM cross compiler
make zynq_zed_defconfig

With the qemu 2.5.0 it recognizes the SD card when you type fatls in u-boot. 
The same u-boot binary does not recognize  SD card  when you use Xilinx qemu.

Its only when you can load the file using fatload, comes the BSP part.

alistair23, 2016-02-23:
I can't reproduce the problem. On both Xilinx QEMU and 2.5.0 QEMU I can't access the SD card from u-boot. Do you mind opening a separate issue and including as much information as possible?

Afra-Felix, 2016-02-29:
Hi Alistair,

Okay, I will create a new issue for this detailing the complete procedure to recreate issue.