Xilinx/qemu #15: Compiler errors

bparker06, 2016-05-19:
Hi, I am trying to compile the master branch on Ubuntu 14.04, but I get lots of these errors after ./configure && make:

In file included from /usr/include/libfdt.h:55:0,
                 from hw/core/fdt_generic_util.c:53:
/usr/include/fdt.h:58:2: error: unknown type name ‘fdt32_t’
  fdt32_t magic;    /* magic word FDT_MAGIC */
  ^
/usr/include/fdt.h:59:2: error: unknown type name ‘fdt32_t’
  fdt32_t totalsize;   /* total size of DT block */
  ^
/usr/include/fdt.h:60:2: error: unknown type name ‘fdt32_t’
  fdt32_t off_dt_struct;   /* offset to structure */
  ^

I also tried using the branch xilinx-v2016.1 and I get a different error there:

  CC    hw/acpi/acpi_interface.o
hw/acpi/pcihp.c: In function ‘acpi_pcihp_pc_no_hotplug’:
hw/acpi/pcihp.c:117:34: error: ‘PCIDevice’ has no member named ‘qdev’
     return (pc->is_bridge && !dev->qdev.hotplugged) || !dc->hotpluggable;
                                  ^
hw/acpi/pcihp.c:118:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
make: *** [hw/acpi/pcihp.o] Error 1

What is the correct way to get qemu to compile?

alistair23, 2016-05-19:
Hello,

Unfortunately due to the vast number of changes that have been made by Xilinx we don't support all of the QEMU targets. It looks like you are trying to compile QEMU for every guest architecture which doesn't work at the moment. We are working hard to improve our code state and be up-to-date with mainline QEMU.

Until that happens you will need to use some config options which are described here: http://www.wiki.xilinx.com/qemu

Thanks,
Alistair

alistair23, 2016-05-19:
Please let me know if that does/doesn't work for you

closed 2016-05-20 by alistair23