Xilinx/qemu #18: qemu-system-arm causes core dumped

JodieChuang, 2016-08-31:
Hi xilinx,

I am using the Xilinx/qemu now. The configuration and compilation are successful. I have built xilinx/qemu/build/aarch64-softmmu and xilinx/qemu/build/arm-softmmu. 
qemu-system-aarch64 works very well, but I failed to use qemu-system-arm, even to check the machine models it supports. 
The output with the error massage is in the following.

~/Documents/xilinx/qemu/build/arm-softmmu$ ./qemu-system-arm -M help
--------------------------------------------------------------------
Xilinx QEMU Aug 24 2016 14:38:39.
--------------------------------------------------------------------
**
ERROR:/home/jiahuan/Documents/xilinx/qemu/qom/object.c:165:type_get_parent: assertion failed: (type->parent_type != NULL)
Aborted (core dumped)

I am running qemu in Linux OS emulated by VirtualBox since my PC is win7.
I have tried with the mainline of Xilinx/qemu and the two active branches as well. All of them produce such an Error when running qemu-system-arm.
I saw an old issue https://github.com/Xilinx/qemu/issues/12 where you also discussed about such a problem. From your wards in this issue, this problem should have been solved, but I don't know why it happened with the latest xilinx/qemu versions.

Any tips will be appreciated.

Best regards,
Jiahuan

JodieChuang, 2016-08-31:
The reason to use qemu-system-arm is that I want to run a 32-bit Linux kernel in qemu.

alistair23, 2016-08-31:
Hello Jiahuan,

The aarch64 QEMU binary includes all of the earlier ARM architectures. It will still let you run a 32-bit kernel. Please use the aarch64 binary instead.
I know it is a little confusing, but this is how mainline QEMU works as well.

I do agree though that this issue needs to be fixed. I'll leave the GitHub issue open until it has been fixed. I am swamped at the moment though so it won't be for at least a week.

Thanks,
Alistair

JodieChuang, 2016-09-01:
Hi Alistair,

I tried the aarch64 xilinx QEMU binary for a 32-bit linux kernel, it does work well.
However, I tried it with the hello world program introduced in https://balau82.wordpress.com/2010/02/28/hello-world-for-bare-metal-arm-using-qemu/ and I didn't get the output string "Hello World".
jiahuan@jiahuan-VirtualBox:~/Documents/xilinx/qemu/build/aarch64-softmmu$ ./qemu-system-aarch64 -M versatilepb -m 128M -nographic -kernel ~/Documents/xilinx/hello_world/test.bin 
--------------------------------------------------------------------
Xilinx QEMU Aug 24 2016 14:38:39.
--------------------------------------------------------------------
audio: Could not init `oss' audio driver

I have checked that the test.elf is 32-bit. 

What will be the reason? 

Regards,
Jiahuan

alistair23, 2016-09-01:
Maybe you haven't connected the correct serial?

I would not recommend using the Xilinx fork of QEMU to run non-Xilinx machines. Our fork has diverged from mainline and we don't test and maintain the non-Xilinx machines.

JodieChuang, 2016-09-05:
Hi, Alistair,

Thank you for your advice.

Regards,
Jiahuan

alistair23, 2017-02-21:
Closing the issue as the core dump is fixed.

closed 2017-02-21 by alistair23

lihaoran8001, 2023-09-19:
If I use aarch64 to run a 32bit image, I cannot debug it throught gdb since the qemu-system-aarch64 set arch to aarch64