Xilinx/qemu #19: Unable to compile using GCC6 (fc24)

rleh, 2016-09-28:
I tried to build xilinxs qemu as described [here](http://www.wiki.xilinx.com/QEMU) and get the following error:
```
[...]
/home/user/xilinx-qemu/hw/intc/arm_gic.c:1322:30: error: ‘gic_cpu_ops’ defined but not used [-Werror=unused-const-variable=]
static const MemoryRegionOps gic_cpu_ops = {
                              ^~~~~~~~~~~
[...]
cc1: all warnings being treated as errors
/home/user/xilinx-qemu/rules.mak:57: recipe for target 'hw/intc/arm_gic.o' failed
make: *** [hw/intc/arm_gic.o] Error 1
make: *** Waiting for unfinished jobs....
```

I am using Fedora 24 (fc24) and gcc from the official repository.
```
$ gcc --version
gcc (GCC) 6.2.1 20160916 (Red Hat 6.2.1-2)
Copyright (C) 2016 Free Software Foundation, Inc.
```

rleh, 2016-09-28:
Adding extra compiler flags `-Wno-error=unused-const-variable` and `-Wno-error=cpp` using the `./configure` script fixes this.
```shell
/configure --target-list="aarch64-softmmu,microblazeel-softmmu" --enable-fdt --disable-kvm --extra-cflags="-Wno-error=unused-const-variable" --extra-cflags="-Wno-error=cpp"
```

alistair23, 2016-09-28:
Hello Raphael,

I have never seen that warning before, I'm not sure why it only appears with GCC6. I will fix the issue and push an update as soon as I can. Unfortunately we are in a freeze here at Xilinx (we are about to release a new version) so it will have to wait until after that.

In the meantime it sounds like you have a solution. This warning won't effect your ability to run QEMU.

For future reference you can also use: --disable-werror to treat all warnings as only warnings and not errors.

Thanks,
Alistair

Bubu, 2017-02-17:
This is still not fixed in master.

alistair23, 2017-02-27:
I pushed a branch that fixes the issue for me, does this work for you?
https://github.com/Xilinx/qemu/tree/build-fixes

alistair23, 2017-03-06:
The fix has been merged into the master branch.
Thanks for reporting the build issue.

closed 2017-03-06 by alistair23