Xilinx/qemu #20: Fails to build with newer glibc (2.49) due to `_BSD_SOURCE` macro

starthal, 2016-09-29:
When building with

    ./configure --target-list="aarch64-softmmu" --enable-fdt --disable-kvm

I get a warning that halts the build:

    In file included from /usr/include/stdint.h:25:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/5/include/stdint.h:9,
                 from /home/salbert/work/hzynq/zynq-vm/qemu/hw/core/remote-port-proto.c:27:
    /usr/include/features.h:148:3: error: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Werror=cpp]
     # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"

I think this has to do with compiling against a newer glibc (2.49.6-1).

Replacing `_BSD_SOURCE` with `_DEFAULT_SOURCE` in `remote-port-proto.c` fixes the build, but I don't know whether there are any side effects.

alistair23, 2016-09-29:
Hello Stephen,

Thanks for reporting the issue. I will look into it.

Thanks,
Alistair

alistair23, 2017-02-27:
I pushed a branch that fixes this issue: https://github.com/Xilinx/qemu/tree/build-fixes

Let me know if that fixes the issue for you.

alistair23, 2017-03-06:
The fix has been merged into master

closed 2017-03-06 by alistair23