Xilinx/qemu #21: Unable to compile using hw/block/xen_disk

B-Bimmermann, 2016-12-08:
Hello,

I have downloaded with:

`git clone git://github.com/Xilinx/qemu.git`

I have install the dependencies:

`sudo apt install libglib2.0-dev libgcrypt20-dev autoconf automake libtool qemu`

I have checked out with:

`git submodule update --init pixman dtc`

When building with:

`./configure --target-list="aarch64-softmmu,microblazeel-softmmu" --enable-fdt --disable-kvm`

I get a warning that halts the build fails:
`  CC    hw/block/pflash_cfi01.o
  CC    hw/block/pflash_cfi02.o
  CC    hw/block/xen_disk.o
hw/block/xen_disk.c: In function ‘ioreq_runio_qemu_aio’:
hw/block/xen_disk.c:557:9: error: implicit declaration of function ‘blk_aio_preadv’ [-Werror=implicit-function-declaration]
         blk_aio_preadv(blkdev->blk, ioreq->start, &ioreq->v, 0,
         ^
hw/block/xen_disk.c:557:9: error: nested extern declaration of ‘blk_aio_preadv’ [-Werror=nested-externs]
hw/block/xen_disk.c:571:9: error: implicit declaration of function ‘blk_aio_pwritev’ [-Werror=implicit-function-declaration]
         blk_aio_pwritev(blkdev->blk, ioreq->start, &ioreq->v, 0,
         ^
hw/block/xen_disk.c:571:9: error: nested extern declaration of ‘blk_aio_pwritev’ [-Werror=nested-externs]
cc1: all warnings being treated as errors
/home/drago/MasterArbeit/qemu/qemu/rules.mak:57: recipe for target 'hw/block/xen_disk.o' failed
make: *** [hw/block/xen_disk.o] Error 1
`
My system is a normal Linux Ubuntu:
`$ uname -a
Linux XXX 4.2.0-42-generic #49-Ubuntu SMP Tue Jun 28 21:26:26 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

`$ gcc --version
gcc (Ubuntu 5.2.1-22ubuntu2) 5.2.1 20151010
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

I have nothing changed and have tried it twice. Every time I get the same error. Must I Ignore the warning ?

best regards,
Benjamin

alistair23, 2016-12-08:
Hello B-Bimmermann,
That is a strange warning, can you try building without Xen? Add --disable-xen when you run configure.
Thanks,
Alistair

B-Bimmermann, 2016-12-08:
Hello Alistair,

when I  build with the config:
`./configure --target-list="aarch64-softmmu,microblazeel-softmmu" --enable-fdt --disable-kvm --disable-xen `
Everything works can be build.

best regards,
Benjamin

alistair23, 2016-12-08:
Great! I'm glad it's working.

Building Xen with QEMU is difficult as the version of Xen has to line-up with the version of QEMU. It also isn't tested by us at Xilinx so it is possible we have introduced a bug in there in our merge process.

You don't need to build QEMU with Xen support (you can still run Xen on QEMU without this). I'm not sure why your system defaults to it being enabled though.
Thanks,
Alistair

closed 2016-12-08 by alistair23