Xilinx/qemu #22: Running qemu-system-aarch64 with xilinx-zynq-a9 results in an error

HotelCalifornia, 2017-02-08:
Running `qemu-system-aarch64 -M xilinx-zynq-a9` always results in the following error, regardless of the other options I specify:

```
Unexpected error in object_property_find() at qom/object.c:995:
qemu-system-aarch64: Property '.has-security-extensions' not found
Aborted (core dumped)
```
I'm trying to simulate a bare metal environment on the xilinx zynq-z7010.

Is there some other way I'm supposed to be doing this? Documentation is a little hazy regarding specific use.

alistair23, 2017-02-08:
Hey Alex,
You are correct, I found the issue and I have some fixes. We have to test them internally before I can push them to master though. I can push a branch to this repo later today though.

Can I recommend that if you want to use the hard coded Xilinx QEMU machines (xilinx-zynq-a9, xlnx-ep108 or xlnx-zcu102) that you use the mainline QEMU project.  We don't add any features to the hard coded boards and we don't really ever run anything on them so they aren't super reliable.

If you want all the extra Xilinx features that we have in our fork you should look at the automatic machine creation. There is more information about that on the wiki: http://www.wiki.xilinx.com/QEMU That is where all of our focus goes and those should always work (our regression tests target those).

Thanks,
Alistair

alistair23, 2017-02-08:
I just pushed a branch that will work for you. It's available here: https://github.com/Xilinx/qemu/tree/xilinx-zynq-a9-fixes

HotelCalifornia, 2017-02-08:
Thank you very much!
I'll test that in a little bit and report back.

alistair23, 2017-02-08:
No worries, sorry it took so long to get back to you.
If you do get a chance to test it and it works for you a Tested-by tag from you would be appreciated. Then I can add that to the commit.

HotelCalifornia, 2017-02-09:
Ran `make install` and got this:
```
~/qemu/hw/cpu/a9mpcore.c: In function ‘a9mp_priv_realize’:
~/qemu/hw/cpu/a9mpcore.c:55:10: error: variable ‘has_el3’ set but not used [-Werror=unused-but-set-variable]
     bool has_el3;
          ^
cc1: all warnings being treated as errors
/home/brookea/qemu/rules.mak:57: recipe for target 'hw/cpu/a9mpcore.o' failed
make[1]: *** [hw/cpu/a9mpcore.o] Error 1
Makefile:186: recipe for target 'subdir-aarch64-softmmu' failed
make: *** [subdir-aarch64-softmmu] Error 2
```
I'll edit the makefile for now, but thought you should know about it anyway

HotelCalifornia, 2017-02-09:
Got it to work by commenting out a few more lines in hw/cpu/a9mpcore.c

alistair23, 2017-02-09:
Hey Alex,
Yeah, I noticed the same warnings later yesterday. An updated patch is currently being reviewed internally and I will push it once it has been reviewed and tested. This process will take longer then usual as we are in a code freeze for our next release at the moment.

Can you comment a Tested-by line so I can include it in the patches?

Thanks,
Alistair

HotelCalifornia, 2017-02-10:
I assume by that you mean in the code, as opposed to here? Should I just add that to the PR I opened?

alistair23, 2017-02-10:
I just added it to the patches. I just pushed the patches, that should fix your issue.

closed 2017-02-10 by alistair23