Xilinx/qemu #24: update to newer qemu?

hpeyerl, 2017-02-20:
I'm in the process of writing a new chardev backend for some custom hardware and the mainline tree has diverged substantially from Xilinx/qemu in this area... Are there any imminent plans to advance the Xilinx/qemu tree closer to mainline?

alistair23, 2017-02-21:
Yeah, we are working on it. We recently updated to being based on 2.6 (which took a very long time) and we are now working on 2.8. The chardev has changed a lot between 2.6 and 2.8 (as I'm sure you know).

Hopefully after the 2017.1 release we can push a 2.8 based branch publicly which you can use. We are always working on keeping up-to-date, it is just a lot of work.

hpeyerl, 2017-02-21:
Thanks.  I wasn't trying to rush you along.  I appreciate the effort put in thus far... Now that I understand the internals of Qemu somewhat, I can see it won't be too difficult to migrate my soft-device to the new chardev world order.

closed 2017-02-21 by hpeyerl

alistair23, 2017-02-24:
It should be pretty easy to do. You can see how we did it with the Xilinx character devices and follow that.