Xilinx/qemu #25: Stalling CPUs in xilinx-qemu but not in qemu-2.8

B-Bimmermann, 2017-07-31:
Hello,

I try to run multiple cortex-a57 cpus. If I use qemu-stable-2.8 everything works fine and if I use only one cortex-a57 with the xilinx-qemu everything also starts. But If I try to start two or more cortex-a57 xilinx-qemu the cpus stalls during the boot.

I can reproduce it every single time with the following steps:

Download xilinx qemu:
`git clone https://github.com/Xilinx/qemu.git`

configure xilinx qemu:
`./configure --target-list=aarch64-softmmu,microblazeel-softmmu,x86_64-softmmu --enable-fdt --disable-pie --disable-brlapi --disable-rdma --disable-rbd --disable-tcmalloc --disable-xen --disable-gtk --disable-glusterfs --disable-xfsctl --disable-kvm --disable-xen --enable-debug --enable-debug-tcg --enable-debug-info`
build xilnx qemu:
`make`

goto the arm system:
`cd aarch64-softmmu/`
get the kernel:
`wget https://www.kernel.org/pub/linux/kernel/v4.x/linux-4.1.39.tar.xz`
unpack the kernel:
`tar -xf linux-4.1.39.tar.xz`
goto the kernel:
`cd linux-4.1.39`
use this config:
[linux-4.1.config .txt](https://github.com/Xilinx/qemu/files/1187398/linux-4.1.qsim-arm64.config.txt)
build the kernel:
`make ARCH=arm64 CROSS_COMPILE =aarch64-linux-gnu- -j4  KCPPFLAGS= "-fno-pic -Wno-pointer-sign" `
go back
`cd..`

This here starts fine:
` ./qemu-system-aarch64 -m 512 -M virt -cpu cortex-a57 -kernel linux-4.1.39/arch/arm64/boot/Image -append "init=/init lpj=34920500 console=ttyAMA0 console=ttyS0 nowatchdog rcupdate.rcu_cpu_stall_suppress=1 console=/dev/ttyS0" -nographic -smp 1`
But with **two** cpus we stall before the linux boot:
`./qemu-system-aarch64 -m 512 -M virt -cpu cortex-a57 -kernel linux-4.1.39/arch/arm64/boot/Image -append "init=/init lpj=34920500 console=ttyAMA0 console=ttyS0 nowatchdog rcupdate.rcu_cpu_stall_suppress=1 console=/dev/ttyS0" -nographic -smp 2`

Maybe this is like the  "target-arm: translate-a64: Fix store exclusive operations" fix. 
[https://github.com/Xilinx/qemu/commit/6ad38bcdbc2c1b532842704b1f21cd089009c6f9](url)

How can we fix that?
I don't know why two or more cpus are stalling.
I have tried to remove every xilinx device in default-configs/arm-softmmu.mak , but the problem still exist.
I think the problem is in the target-arm/translate.c. 

I would like to get an instruction about what it can be, or how I can fix it up.
Have a nice day

alistair23, 2017-08-07:
Hello B-Bimmermann,

Why are you using the Xilinx fork of QEMU to run non-Xilinx devices? We don't spend any effort testing or maintaining non Xilinx devices in our fork. If the mainline release of QEMU works why don't you just use that?

B-Bimmermann, 2017-08-11:
Hello alistair23,

in our research work we evaluate multiple ARM CPUs versus the ZCU102 with and without hardware acceleration. In order to make it comparable, we want to use the same QEMU machine as far as possible. Because all our test cpus and all xilinx cpus and devices are possible in the aarch64-softmmu.mak config we want to use this config as a basis for our test. Because of the xilinx patches for the ZCU102 and the hardware systemC acceleration, we are forced not to use the standard qemu. 

We tried to use the Xilinx kernel at the beginning, however, we did not manage to get it started under qemu. That's why we use the mainline kernel at the moment. If you have some advice for us for the ZCU102 with the virt machine, we would be glad. 

Continuing our tests we got the same stalling for linux with the cortex-a53. 
As the ZCU102 has at least two cortex-a53 cpus, it should work in xilinx qemu. 
To reproduce the error:
`./qemu-system-aarch64 -m 512 -M virt -cpu cortex-a53 -kernel linux-4.1.39/arch/arm64/boot/Image -append "init=/init lpj=34920500 console=ttyAMA0 console=ttyS0 nowatchdog rcupdate.rcu_cpu_stall_suppress=1 console=/dev/ttyS0" -nographic -smp 2`

For these reasons listed above we think the xilinx qemu should be patched to fix this bug.

I would like to get an instruction about what it can be, or how we can fix it up.
Have a nice day

alistair23, 2017-08-11:
I'm not sure what you are trying to evaluate with QEMU, but I don't think QEMU is the best option for that. It is more targeted towards development and debugging, but that choice is up to you.

As for the issue you are seeing without images and logs there is nothing we can do. The Xilinx kernel boots on the Xilinx QEMU, but you will need to use the Xilinx specific machines as described on the wiki (http://www.wiki.xilinx.com/QEMU). If you are trying to boot a ZCU102 kernel on the virt machine I really doubt it will ever work as all the ZynqMP specific devices are missing.

Unfortunately we don't have any spare QEMU resources and we can't spend any time fixing bugs on non-Xilinx machines. In saying that if you do supply images, sets to rebuild the same images and logs I might be able to have a quick look.

B-Bimmermann, 2017-08-14:
Hello alistair23,

first of all thanks for your answer. To your first statement:
Only QEMU is fully supported by Xilinx and so on only in QEMU there are most XILINX devices.
Thats the primary reason why we must use QEMU.

To the Kernel: Is it possible to boot just the kernel and the file-system without the boot loader in the arm-generic-fdt?
Are there any other machines that are supported by Xilinx, apart from arm-generic-fdt?

A second problem: (should I open a second issue ?)
We have now upgraded from Petalinux 2016.4 to 2017.2.
The wiki [(http://www.wiki.xilinx.com/QEMU)](url) is outdated. You should upgrade it.
We can start the Kernel with :

```
# telnet adress
MON_ADDR="127.0.0.1"
# random port number for monitoring port
MON_PORT=$( shuf -i 10000-64000 -n 1)
GDB_PORT=$( shuf -i 10000-64000 -n 1)

DIR_QEMU="/your_path_to/ [https://github.com/Xilinx/qemu] "
DIR_DEVICE_TREES_XILINX=""/your_path_to/ [https://github.com/Xilinx/qemu-devicetrees.git] /LATEST/SINGLE_ARCH"
DIR_PETALINUX_ZCU102_PROJECT="/your_path_to//xilinx-zcu102-zu9-es2-rev1.0-2017.2"

# Set the VM Name
VM_NAME="QEMU_XILINX_XCU102_ARM"
# Set the Path to the QEMU SYSTEM
VM_SYSTEM_DIR_BASE="$DIR_QEMU/aarch64-softmmu/qemu-system-aarch64 "

VM_CMD_LINE=" \
-name $VM_NAME \
-nographic \
-no-reboot \
-localtime \
-monitor telnet:$MON_ADDR:$MON_PORT,server,nowait \
-serial mon:stdio \
-serial /dev/null \
-display none \
-gdb tcp::$GDB_PORT \
 \
-M arm-generic-fdt \
-device loader,addr=0xfd1a0104,data=0x8000000e,data-len=4 \
-global xlnx,zynqmp-boot.cpu-num=0 \
-global xlnx,zynqmp-boot.use-pmufw=true \
-device loader,file=$DIR_PETALINUX_ZCU102_PROJECT/images/linux/bl31.elf,cpu-num=0 \
-device loader,file=$DIR_PETALINUX_ZCU102_PROJECT/images/linux/Image,addr=0x00080000 \
-device loader,file=$DIR_PETALINUX_ZCU102_PROJECT/images/linux/system.dtb,addr=0x1407f000 \
-device loader,file=$DIR_PETALINUX_ZCU102_PROJECT/build/misc/linux-boot/linux-boot.elf \
-dtb $DIR_PETALINUX_ZCU102_PROJECT/images/linux/system.dtb
 \
-hw-dtb $DIR_DEVICE_TREES_XILINX/zcu102-arm.dtb \
-pflash $DIR_PETALINUX_ZCU102_PROJECT/images/linux/nand0.qcow2 -m 4G \
 "
VM_START_LINE="$VM_SYSTEM_DIR_BASE $VM_CMD_LINE"

echo "[INFO] QEMU command line: $VM_START_LINE"
eval $VM_START_LINE
```
The device-tree position has changed from 0x04080000 to 0x1407f000 and we have to add the global args. I have sent an E-Mail with this error also to veena@xilinx.com.
But the Kernel stops booting:

```
audio: Could not init `oss' audio driver
NOTICE:  ATF running on XCZUUNKN/QEMU v1/RTL0.0 at 0xfffea000
NOTICE:  BL31: Secure code at 0x60000000
NOTICE:  BL31: Non secure code at 0x8000000
NOTICE:  BL31: v1.3(release):0d9d51a
NOTICE:  BL31: Built : 10:39:41, Aug 11 2017
[    0.000000] Booting Linux on physical CPU 0x0
[... 28 lines cut ...]
[    0.000000]        .bss : 0xffffff80091dda00 - 0xffffff800923ff2c   (   394 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf00000000 - 0xffffffbf1dc00000   (   476 MB actual)
[    0.000000]     memory  : 0xffffffc000000000 - 0xffffffc880000000   ( 34816 MB)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 64.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=64, nr_cpu_ids=4
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] GIC: Adjusting CPU interface base to 0x00000000f902f000
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 50.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
[    0.000580] sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
[    0.037042] Console: colour dummy device 80x25
[    0.037691] console [tty0] enabled
[    0.039588] bootconsole [cdns0] disabled

```

Do you need any more logs? If yes witch one? Every system is installed and compiled like in the wiki. We have used the xilinx-zcu102-zu9-es2-rev1.0-v2017.2-final for petalinux. With the new petalinuxboot the device tree is every time zynqmp-qemu-multiarch-arm.dtb but we would prefer to boot just the kernel and the file system without the SystemC integration.

Thanks a lot

alistair23, 2017-08-14:
I guess it would be technically possible to boot directly into the kernel, but it would require a lot of work from your side. I'm not sure why you would want to.

You can use any machine you want as long as it accurately models the hardware that your guest expects. As I said earlier Xilinx only supports models of Xilinx boards. For ZynqMP arm-generic-fdt is your best bet.

Thanks for pointing out the wiki issue, we will update it.

If you are using PetaLinux 2017.1 or newer you need to use the PMU firmware as well. Please use petalinux-boot --qemu to do this.

B-Bimmermann, 2017-08-15:
Hello alistair23,

`If you are using PetaLinux 2017.1 or newer you need to use the PMU firmware as well. Please use petalinux-boot --qemu to do this.`

if I'm now correct for `petalinux 2017.1` and later it is impossible to start xilinx-qemu with any SINGLE_ARCH device tree ?
Please confirm me that for `petalinux 2017.1` and later the ZCU102 every startup MUST be a MULTI_ARCH start, because we must start the `qemu-system-aarch64 -M arm-generic-fdt`
 and the `qemu-system-microblazeel -M microblaze-fdt`  which have the pmu.

If that is correct the wiki for `Running a Zynq UltraScale+ Linux Kernel Image On Xilinx's ARM QEMU` is outdated and can't be run or ?

alistair23, 2017-08-15:
As of PetaLinux 20171. all of the PetaLinux builds require the PMU FW. So it is impossible to start those binaries with just the single arch device trees.

The wiki is not outdated as you can still use QEMU to run just the ARM instance. If you wanted to remove the requirement on PMU FW from PetaLinux or wanted to boot a mainline kernel you don't need the PMU FW and we include steps for running these images on QEMU.

It is up to the user to determine what they need to run their images. PetaLinux will also do this for you.

B-Bimmermann, 2017-08-15:
Hello alistair23,

First, I wanted to tell you that I am very glad for your help. Can you give me a link, where I find the WIKI or a guide how I build petalinux 2017.2 without the PMU, but with the Xilinx-Kernel to run just the ARM instance. Maybe I have missed something.

Thanks a lot.

alistair23, 2017-08-15:
Unfortunately I cannot. I am just assuming it is possible, but I don't have any steps to do it. You will have to figure that out yourself or contact Xilinx support.

I'm sorry I can't be more helpful but that is well out of the scope of QEMU.

I am going to close this issue now. If you come across any Xilinx QEMU problems in the future feel free to open a new issue.

closed 2017-08-15 by alistair23

alistair23, 2017-08-15:
I have also updated the wiki steps to match the latest PetaLinux release and make the PMU FW requirements more explicit.