Xilinx/qemu #26: _DEFAULT_SOURCE redefined

MedicineYeh, 2017-11-03:
Hi, 
On my system (the latest release of ArchLinux), config-host.mak contains the same flag and would cause a redefinition of this macro. The following is my fix and perhaps you can apply this patch to your master branch.

Compiler: g++ (GCC) 7.2.0

``` patch
From 58cb86a73af85ed7be12f6f5edc56bdcb76f9d4a Mon Sep 17 00:00:00 2001
From: Medicine Yeh <medicinehy@gmail.com>
Date: Fri, 3 Nov 2017 12:20:43 +0800
Subject: [PATCH 2/2] Fix _DEFAULT_SOURCE redefined in Xilinx QEMU

---
 hw/core/remote-port-proto.c | 2 ++
 1 file changed, 2 insertions(+)

diff --git a/hw/core/remote-port-proto.c b/hw/core/remote-port-proto.c
index 4fc29ed..a2c7946 100644
--- a/hw/core/remote-port-proto.c
+++ b/hw/core/remote-port-proto.c
@@ -23,7 +23,9 @@
  * THE SOFTWARE.
  */
 
+#ifndef _DEFAULT_SOURCE
 #define _DEFAULT_SOURCE
+#endif
 #include <stdint.h>
 #include <stdlib.h>
 #include <stdio.h>
-- 
2.14.2
```

alistair23, 2017-11-06:
Thanks so much for the patch!

Can you create a pull request with a patch following the QEMU guidelines (https://wiki.qemu.org/index.php/Contribute/SubmitAPatch)?

If you have any questions just as here or submit a pull request and I can help walk you through anything.

alistair23, 2018-01-08:
Closing this as it was fixed by someone else: https://github.com/Xilinx/qemu/commit/a886a3f26bd225e649dc3a0f589dce193ee9a6ea

closed 2018-01-08 by alistair23