Xilinx/qemu #28: QEMU can't realize the axi dma node

bsafwen, 2018-04-03:
I am having an issue  emulating my board, which is base on the zynq-700 board series. When I run qemu, it segfaults and dumps a core file, here is the [backtrace](https://pastebin.com/FMHA4kJh).

I am using the 2017.3 tag for both the kernel and qemu. and here is [my device tree](https://pastebin.com/a1vGk8EB).

The command I use to launch qemu:
`/root/qemu/aarch64-softmmu/qemu-system-aarch64 -M arm-generic-fdt-7series -machine linux=on -serial /dev/null -serial mon:stdio -nographic -dtb dtree.dtb -kernel /root/linux-xlnx/arch/arm/boot/zImage -drive if=sd,format=raw,index=0,file=/root/rootfs/sd.img -append "root=/dev/mmcblk0 rw rootwait rootfstype=ext2"
`

alistair23, 2018-04-03:
Hey Safwen,

You are seeing the seg fault because your DMA node (dma@40400000) doesn't have the sg property set. You need to ensure that the sg property in the device tree points to a valid QEMU MemoryRegion (https://github.com/Xilinx/qemu/blob/master/hw/dma/xilinx_axidma.c#L609).

Have a look at a Xilinx provided DTS and you can just copy it from there.

bsafwen, 2018-04-05:
Hi @alistair23, thank you for your answer, it no longer segfaults, but now I am stuck at 
`Waiting for root device /dev/mmcblk0...` and then I there is this debug message "Skipping voltage switch". After some debugging, I found that this functions does not succeed:
`linux-xlnx/drivers/mmc/core/sd.c: int mmc_sd_get_cid(struct mmc_host *host, u32 ocr, u32 *cid, u32 *rocr)`
 and this function calls 

```
> err = mmc_set_signal_voltage(host, MMC_SIGNAL_VOLTAGE_180,
>                                         pocr);
>                 if (err == -EAGAIN) {
>                         retries--;
>                         goto try_again;
>                 } else if (err) {
>                         retries = 0;
>                         goto try_again;
>                 }
```
here err is 25, and so this function never returns because it could not set the voltage signal.
A stupid workaround that I did, was to set err to 0 just before the check and now the kernel can load the rootfs. But it took me 5 minutes to get to the prompt login, issuing a command like ls takes at least 10 seconds.

Thanks @alistair23

alistair23, 2018-04-05:
Hey Safwen,

I don't see anything obviously wrong with what you are doing, it all looks standard in which case QEMU should not take that long to boot or to run commands. Maybe try not an SD boot (JTAG boot) to just narrow down the issue?

Unfortunately I can't help reproduce and fix the problem, you will need someone from Xilinx to help you there.

bsafwen, 2018-04-06:
@alistair23 ok thank you, I will reach out to someone from Xilinx. thanks.

closed 2018-04-06 by bsafwen