Xilinx/qemu #30: Qemu segmentation fault on Ubuntu

t6george, 2018-07-31:
Hello,

I am trying to run QEMU (this repository) on an arm-generic-fdt-7series. My goal for now is to et to the bootloader, but when I run the following command:

./aarch64-softmmu/qemu-system-aarch64 -M arm-generic-fdt-7series -serial mon:stdio -serial /dev/null -display none -device loader,addr=0xfd1a0104,data=0x8000000e,data-len=4 -device loader,file=/home/tgeorge/QEMU/qemu/xilinx2016.2/zcu102/bl31.elf,cpu-num=0  -device loader,file=/home/tgeorge/QEMU/qemu/xilinx2016.2/zcu102/u-boot.elf -hw-dtb /home/tgeorge/QEMU/qemu/xilinx2016.2/zcu102/system.dtb

I get a segmentation fault (core dumped). Is this an issue with my command (I got from http://www.wiki.xilinx.com/QEMU+-+Zynq+UltraScalePlus), or is this an issue with the installation/bug?

Thank you for your help!

t6george, 2018-07-31:
Update: I changed the device tree, and now I am getting:

audio: Could not init `oss' audio driver
qemu-system-aarch64: qemu_fdt_get_phandle: Couldn't get phandle for /lmb_pmu@0/memory@ffd00000: <no error>

Perhaps it is this file that is causing the problem. Where can I get the DTB for ZynqMP?

alistair23, 2018-07-31:
You can get the latest QEMU device tree from: https://github.com/Xilinx/qemu-devicetrees.

It is best to use the latest QEMU and device trees as they are tested together.

t6george, 2018-07-31:
Thank you, however I still get the same error message as above

alistair23, 2018-07-31:
Can you attached the backtrace of the seg fault?

t6george, 2018-07-31:
I switched from using the system.dtb to zynqmp-pmu.dtb in my command:

./aarch64-softmmu/qemu-system-aarch64 -M arm-generic-fdt -serial mon:stdio -serial /dev/null -display none -device loader,addr=0xfd1a0104,data=0x8000000e,data-len=4 -device loader,file=./xilinx2016.2/zcu102/bl31.elf,cpu-num=0  -device loader,file=./xilinx2016.2/zcu102/u-boot.elf -hw-dtb ../qemu-devicetrees/LATEST/SINGLE_ARCH/zynqmp-pmu.dtb

and I still get the error message:
audio: Could not init `oss' audio driver
qemu-system-aarch64: qemu_fdt_get_phandle: Couldn't get phandle for /lmb_pmu@0/memory@ffd00000: <no error>

(no more seg fault).

I am using the up-to-date device tree from GitHub.

Some documentation references ./pre-built/linux/images/zynqmp-qemu-arm.dtb, but I do not have that file or directory. Was this file replaced with another?

alistair23, 2018-07-31:
That is definitely the wrong DTB to use. I just re-read your first post and it seems like you are using the wrong HW device tree there as well. Can you go through the documentation again and use the DTB you built from https://github.com/Xilinx/qemu-devicetrees for the hardware DTB

bsreedha, 2018-10-12:
Alistair, what is the complete command line for bringing up U-boot with Xilinx QEMU on  arm-generic-fdt-7series? I have been trying different variations of this - 
./qemu-system-aarch64 -M arm-generic-fdt-7series -serial null -serial mon:stdio -machine linux=off -display none 
-device loader,addr=0xf8000008,data=0xDF0D,data-len=4 
-device loader,addr=0xf8000140,data=0x00500801,data-len=4 
-device loader,addr=0xf800012c,data=0x1ed044d,data-len=4 
-device loader,addr=0xf8000108,data=0x0001e008,data-len=4 
-device loader,file=/path/to/my/2016.4-zc702-release/fsbl-zc702-zynq7.elf,cpu-num=0 
-device loader,file=/path/to/my/2016.4-zc702-release/u-boot.elf 
-dtb /path/to/my/zynq-zc702.dtb 
-D qemu.log -d in_asm

Is that correct? I seem to be hanging at - 
IN: main
0x000000000000ce2c:  ebffcdf2      bl	0x5fc

----------------
IN: FsblHookFallback
0x00000000000005fc:  eafffffe      b	0x5fc

alistair23, 2018-10-12:
I no longer work for Xilinx so I can't really be of much help here. Your QEMU command line looks fine to me, but I'm not certain. It's entirely possible that FSBL or u-boot is broken and you need to make changes to it to run on QEMU. You will have to ask Xilinx for help with booting it unfortunately.

When doing that it's important that you include version information for all of your images, including QEMU and the device trees. It is also important that all these versions line up.

bsreedha, 2018-10-12:
Thanks for looking into it. I will ask someone from Xilinx.