Xilinx/qemu #34: Zynq SMP 2

sangorrin, 2019-02-05:
Hi,

A few years ago this repository's hw/arm/xilinx_zynq.c supported multi-core (2 cores). Now it seems to only support 1 core.

$ qemu-system-arm -M xilinx-zynq-a9 -smp 2 \
		-serial vc:80Cx40C -serial vc:80Cx40C -serial vc:80Cx40C -serial vc:80Cx40C \
		-no-reboot -icount auto -m 1024M -kernel ../fmp_1.4.0/obj/obj_sample/fmp.bin
	-> qemu-system-arm: Invalid SMP CPUs 2. The max CPUs supported by machine 'xilinx-zynq-a9' is 1

Was there any specific reason to remove that support?
Would it be hard to restore it?
I can help if you give me a short overview of the problem.

Thanks,
Daniel

closed 2022-12-22 by sangorrin

hcastaing, 2024-07-02:
Hello, 
Is it possible to share the reason why this is not implemented ? In my understanding, according to https://qemu.readthedocs.io/en/master/system/arm/xlnx-zynq.html QEMU supposedly support the CPU with 2 cores.

opticron, 2024-10-29:
I looked through the git history and apparently the xilinx-zynq-a9 machine model never actually supported SMP in QEMU and I can't find references to the documentation where that claim is made.