Xilinx/qemu #39: Problems is fdt_init_node (fdt_generic_util.c)

cmcoldwell, 2020-01-02:
Testing the variable compat_len before it is initialized.
Leaking memory (device_type = g_strdup_printf("device_type:%s", device_type);

My gawd: this is bush league stuff.

I'm working on a patch.

cmcoldwell, 2020-01-04:
I have a set of patches to propose.
[0001-Use-a-standard-linked-list-idiom.txt](https://github.com/Xilinx/qemu/files/4021840/0001-Use-a-standard-linked-list-idiom.txt)
[0002-Straighten-out-the-logic-in-fdt_init_node.txt](https://github.com/Xilinx/qemu/files/4021841/0002-Straighten-out-the-logic-in-fdt_init_node.txt)
[0003-Fix-segfault-in-xilinx_intc_fdt_auto_parent.txt](https://github.com/Xilinx/qemu/files/4021842/0003-Fix-segfault-in-xilinx_intc_fdt_auto_parent.txt)
[0004-Put-sysbus-MMIO-in-system-memory-map.txt](https://github.com/Xilinx/qemu/files/4021843/0004-Put-sysbus-MMIO-in-system-memory-map.txt)

edgarigl, 2020-01-08:
Thanks for contributing!

A few comments:

Unfortunately the patches do not pass checkpatch, you can try on your own by doing:
git show <git-commit-id> | ./scripts/checkpatch.pl -

It would also be great if you could prefix the summary of the commit message with something.
E.g fdt-generic: Some description
You can see examples in the history for a specific file or submodule with git log -- file.c

Also, if the patch is non-obvious, it would be great if you could provide some context in the commit-message, e.g for patches 2 & 4.

I've fixed up and applied patch 1 & 3.

Could you please fix up and resubmit patch 2 & 4, perhaps as a pull request on github?

Thanks again for contributing!

Cheers,
Edgar

cmcoldwell, 2020-01-08:
> git show | ./scripts/checkpatch.pl -

I went over the other two patches and expanded on my justification in the
commit logs, and made sure they pass they scripts/checkpatch.pl test. I
should warn that it is possible that 634b8af2ad67 introduces a deadlock if
there is ever a CPU waiting for the interrupt controller to initialize
(while the interrupt controller is waiting for the CPU to initialize). I
don't think the situation arise in practice, but I would feel better if
there were some kind of regression test. It does, however, work for me and
fixes a segfault.

My last patch "Put sysbus MMIO in system-memory map" is potentially
controversial also, if there are places where the sysbus registers
shouldn't go into the system-memory (global) memory map.

cmcoldwell, 2020-01-08:
Forgot to add the patches.

cmcoldwell, 2020-01-08:
[0001-Straighten-out-the-logic-in-fdt_init_node.txt](https://github.com/Xilinx/qemu/files/4035586/0001-Straighten-out-the-logic-in-fdt_init_node.txt)
[0002-Put-sysbus-MMIO-in-system-memory-map.txt](https://github.com/Xilinx/qemu/files/4035587/0002-Put-sysbus-MMIO-in-system-memory-map.txt)

edgarigl, 2020-01-08:
Hi,

I've applied patch #1. 

Patch number 2 breaks our system though. We rely quite heavily on attaching devices to the actual parent bus. I'm not sure how we'd work around this for device-tree's that don't match this. Perhaps you could find a way to specifically catch the orphaned case and apply your logic only in that case.

Did you publish the device-tree causing problems somewhere?
Is that device-tree auto-generated or are we in a position to hand-edit it?

Thanks,
Edgar

cmcoldwell, 2020-01-08:
The device tree that causes the problem was auto-generated by the Xilinx HSI utility (although I hand edited system-top.dts to add the memory@0 node). I'm attaching the files.

I'll take another look and see if I can find a better way. To be honest, this method felt pretty heavy-handed when I did it.

[dts.zip](https://github.com/Xilinx/qemu/files/4036712/dts.zip)

cmcoldwell, 2020-01-10:
OK, here's another swing at the problem. I'll admit my understanding of QOM and QDev (and the relationship between the two) is limited, but it looks like when MMIO regions (e.g. "amba_pl") are "orphaned", they get added as "child<simple-bus>" type properties to the root node of the object tree. This code finds those properties and adds them to the system bus memory map. Passes checkpatch.pl.

[0001-Put-orphaned-MMIO-into-system-memory-map.txt](https://github.com/Xilinx/qemu/files/4047794/0001-Put-orphaned-MMIO-into-system-memory-map.txt)

cmcoldwell, 2020-01-16:
Any feedback on this patch?

https://github.com/Xilinx/qemu/issues/39#issuecomment-573193709