Xilinx/qemu #40: ERROR: glib-2.48 gthread-2.0 is required to compile QEMU

alvaro-leite, 2020-02-26:
I'm trying to follow your installation tutorial, but it fails when I run

`./configure --target-list="aarch64-softmmu,microblazeel-softmmu" --enable-fdt --disable-kvm --disable-xen`

The error thrown in the terminal is 

`ERROR: glib-2.48 gthread-2.0 is required to compile QEMU`

At this point, I'm having trouble reverting my system to those versions.

edgarigl, 2020-02-26:
Hi Alvaro,

You don't need those exact versions but you need to install the -dev packages, e.g apt-get install libglib2.0-dev.

Best regards,
Edgar

closed 2020-02-26 by alvaro-leite

PoplarYang, 2020-07-08:
for CentOS
```bash
# yum install glib2
# yum install glib2-devel
```

abitofmaya, 2020-07-08:
I have installed most of the dependencies below. I am still getting this error.
```
sudo apt install libglib2.0-dev libgcrypt20-dev zlib1g-dev autoconf automake libtool bison flex libpixman-1-dev libnfs-dev libiscsi-dev libaio-dev libbluetooth-dev libbrlapi-dev libbz2-dev libcap-dev libcap-ng-dev libcurl4-gnutls-dev libgtk-3-dev

../configure --target-list="aarch64-softmmu,microblazeel-softmmu" --enable-debug --enable-fdt --disable-kvm --disable-xen
```

penpendandan, 2020-08-01:
> for CentOS
> 
> ```shell
> # yum install glib2
> # yum install glib2-devel
> ```
however， I have glib2.4，and the error still remain.
I run
`yum install glib2`
the log in the terminal  is 
`Package glib2-2.40.0-4.x86_64 already installed and latest version`

I run
`yum install glib2-devel`
the log in the terminal  is 
`Package glib2-devel-2.40.0-4.x86_64 already installed and latest version`

How can I fix it?

avisWang, 2021-03-30:
you have already installed glib2 and glib2-devel, 
sudo ./configure
sudo make
that works for me

Diaz-Manuel, 2021-09-12:
> 
> 
> I am using Ubuntu 20.04. I cannot solve this problem.
> 
> Here is my error log.
> 
> ```shell
> ❯ ./configure --target-list=riscv64-softmmu && make
> Using './build' as the directory for build output
> ERROR: glib-2.56 gthread-2.0 is required to compile QEMU
> ```
> 
> I've already installed `libglib2.0-dev` and cannot find gthread. What should I do?

After installing the package `libglib2.0-dev` (together with `libpixman-1-dev` due to later errors) I was able to run `sudo ./configure` errorless. Just wouldn't work without sudo.

samuelcolqueflores, 2022-07-01:
I had a totally different experience to say that the error `ERROR: glib-2.48 gthread-2.0 is required to compile QEMU` is ambiguous.
I did have the package `libglib2.0-dev` installed, so the error was not related to it.
After inspecting the `config.log` file (a file in one's build directory), I discovered that some libraries and headers where missing because the compilation failed after adding a library or because of an undefined function. Some libraries were (in my case) namely: libibumad, libibverbs. Thus, I installed `libibumad-dev`, `libibverbs-dev` packages in my system (Ubuntu).
The missing libraries or headers in your case may vary so inspect the lasts lines of the `config.log` file in your build directory to see which libraries/headers you don't have installed.

wang-jiaolong, 2022-08-30:
> I am using Ubuntu 20.04. I cannot solve this problem.
>
> Here is my error log.
>
> ```shell
> ❯ ./configure --target-list=riscv64-softmmu && make
> Using './build' as the directory for build output
> ERROR: [...]
I have also encountered this problem. Have you solved it?

saipava, 2022-08-30:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Hi Jialong,

As you are on ubuntu. Did you ever try

 apt build-dep qemu

Regards,
Sai Pavan

wang-jiaolong, 2022-08-30:
Thank you for your advice, but after downloading some files, the problem is still unsolved and the same error message remains.

neonb88, 2022-11-29:
> 

wait but did it work after you did this?

mrsirfloppa, 2023-02-23:
here's my error:
 ./configure
Using './build' as the directory for build output

ERROR: glib-2.56 gthread-2.0 is required to compile QEMU