Xilinx/qemu #41: Cosimulation Baremetal

nvgubba, 2020-04-17:
Hi,

Can you please provide QEMU Cosimulation + SystemC examples for Baremetal code, Xilinx Cosimulation instructions show only Linux Cosimulation. I am very much interested to know how this works with Baremetal or any RTOS

edgarigl, 2020-04-17:
Hi,

It works the same with baremetal as with Linux. For QEMU and the SystemC simulation both are just SW.

What chip do you want to emulate? Zynq-7000 or a ZynqMPSoC?

nvgubba, 2020-04-17:
Hi,
Anychip is fine for me for emulation. 
I am curious how to invoke the QEMU and SystemC Cosimulation when it is Barmetal. 
1. Precicesly i am looking whether hw-dtb is required during Baremetal Simulation, what will be the qemu arguments during that time
2. Also, i want to see atleast one read and write transaction goes from QEMU to SystemC periperhal.

If you can provide this it will be very helpful. I searched in the internet and xilinx forumt to cosimulate baremetal, unfortunately i didn't get any good pointers

edgarigl, 2020-04-17:
Yes, the -hw-dtb option is always needed. The hw-dtb file describes the hardware that QEMU will emulate.

The -dtb option is not needed though.

Here's a ZynqMPSoC example from our testsuite, you'll have to put in your own baremetal.elf:

qemu-system-aarch64 -M arm-generic-fdt -hw-dtb /home/edgari/build-dts/LATEST/SINGLE_ARCH/zcu102-arm.cosim.dtb -serial stdio -display none -device loader,file=baremetal.elf,cpu-num=0 -device loader,addr=0xfd1a0104,data=0x8000000e,data-len=4  -machine-path /tmp/machine-path/

zynqmp_demo unix:/tmp/machine-path/qemu-rport-_amba@0_cosim@0 10000

nvgubba, 2020-10-13:
Hi,

Thanks for the feedback

closed 2020-10-13 by nvgubba