Xilinx/qemu #42: RISCV Cosimulation

nvgubba, 2020-05-06:
Hi,
I want to try RISCV QEMU Cosimulation running baremetal RISCV application running on QEMU . Could you please provide instructions or user guide for Baremetal RISCV cosimulation? Thanks

edgarigl, 2020-05-06:
https://github.com/Xilinx/systemctlm-cosim-demo/blob/master/docs/lmac-demos.md

nvgubba, 2020-05-06:
Hi,
This below command is still applicable or do i need to change any arguments for running baremetal

`sudo qemu-system-riscv64 -nographic -machine virt \
     -kernel riscv-pk/build/bbl -append "root=/dev/vda ro console=ttyS0" \
     -drive file=busybear-linux/busybear.bin,format=raw,id=hd0 \
     -device virtio-blk-device,drive=hd0`

edgarigl, 2020-05-06:
The TBM repo has examples on bare-metal for the QEMU virt and ariane test-harness machine:
https://github.com/edgarigl/tbm

For example, try:
qemu-system-riscv64 -M virt -m 256M -serial stdio -display none \
        -device loader,file=baremetal.elf,cpu-num=0

baremetal.elf being your ELF targeting the QEMU virt machine (similar address-space as ariane test harness machine).
By replacing -M virt with -M virt-cosim you'll get the remote-port SystemC connections that work with the systemctlm-cosim-demo. We don't have an explicit riscv64_demo in systemctlm-cosim-demo yet but you should be able to use the zynqmp_demo in the meantime.

Cheers,
Edgar

nvgubba, 2020-05-08:
Hi Edgar,
Thanks for the information. 

I have followed the above instructions below is the output, but i dont see any transactions go to the SystemC side. Am, i missing somthing here. It will be a great help, if i see some transactions going to SystemC side.
`./qemu-system-riscv64 -M virt-cosim -m 256M -serial stdio -display none -device loader,file=/home/rajesh/work/tbm/build/ariane_soc/ctest-bare,cpu-num=0
qemu-system-riscv64: warning: No -bios option specified. Not loading a firmware.
qemu-system-riscv64: warning: This default will change in a future QEMU release. Please use the -bios option to avoid breakages when this happens.
qemu-system-riscv64: warning: See QEMU's deprecation documentation for details.

TBM May  8 2020 08:31:50 running on ariane_soc-
** ariane_soc-: Run CRT tests (may run forver)
CRT: All done.

	- halt

[1]+  Stopped                 ./qemu-system-riscv64 -M virt-cosim -m 256M -serial stdio -display none -device loader,file=/home/rajesh/work/tbm/build/ariane_soc/ctest-bare,cpu-num=0
`

edgarigl, 2020-05-08:
Hi,

Yes, you'll have to edit the baremetal code or provide your own that makes transactions to the SystemC side. You'll also need to edit the addresses in the zynqmp_demo to match.
You can look at the riscv_lmac_demo to see the base addresses.

I'll try to add some better examples one of these days when I find some time.

Best regards,
Edgar

nvgubba, 2020-05-18:
Hi Edgar,
You mentioned you will add some better examples. Will it possible to give some examples?

Best regards,
Rajesh

edgarigl, 2020-05-19:
Hi Rajesh,

Yes, I've got it on my TODO list but I'm currently stuck with some other work. I'll reply to this thread when I've added something. Hopefully end of this week or early next week.

Cheers,
Edgar

nvgubba, 2020-06-03:
Hi Edgar,

I am trying to run the qemu with systemc lmac2 demo. Sometimes Qemu is showing transactions exceeded and qemu stops simulation. May i know what is the reason, whether systemC lmac2 demo is not able to cope up the transactions from the Qemu world

Following are the steps i followed.

`root@busybear:~# ifconfig eth1 up
qemu-system-riscv64: Number of outstanding transactions exceeded! 32
qemu-system-riscv64: /machine/cosim: Internal error clk=155208819861 ns`

Beset Regards
Rajesh

nvgubba, 2020-10-13:
Closing the issue, i am not facing issues now

closed 2020-10-13 by nvgubba

CHINTADINESH, 2021-10-10:
> Hi Edgar,
>
> I am trying to run the qemu with systemc lmac2 demo. Sometimes Qemu is showing transactions exceeded and qemu stops simulation. May i know what is the reason, whether systemC lmac2 demo [...]

Hi @nvgubba,

I am getting the same error as you did. From application standpoint, I am trying to copy an array of integers using memcpy to one of the memory mapped IO addresses. 

It would be really helpful if you could share your experience in fixing the issue.

Thanks,
Dinesh

pedduf, 2022-01-17:
Hi, 
I would like to know how it is possible to access the remote port in the RISCV co-simulation demo and make a data transfer between RISCV qemu and systemc through remote port without using LMAC. Do I have to add a remote port in the Devicetree file ? 
If yes, is there any example on how to do that ?  
Thank you in advance for your help. 
Best Regards
Ped

pedduf, 2022-01-18:
> Hi Ped,
> 
> I'm not 100% sure what you mean but the easiest way to test things out is by using Linux devmem to do physical memory accesses from a userspace shell. You don't need anything in the device-tree for that but you will need to make your the SystemC side has some device connected on the address you're trying to access.
> 
> Best regards, Edgar

thank you edgar for the reply. Basically we need a memory mapped based mechanism for data transfer between Qemu and Systemc part. As I have already seen the cosimulation document from Xilinx, there is a remote port module in QEMU to enable this mechanism. IS there such a thing in RISCV cosimulation demo ? 
Best Regards, 
Pedram

edgarigl, 2022-01-18:
Yes, the same remote-port communication mechanism is used in the RISCV demo.
On the riscv demo, the remote-port devices are instantiated from C code in QEMU:
https://github.com/Xilinx/qemu/blob/master/hw/riscv/virt.c#L525

As opposed to the ZynqMP demo, where the remote-port nodes are created in the Xilinx Hardware device-tree.

Cheers,
Edgar

pedduf, 2022-01-25:
thanks.  I look for a memory mapp based data exchange scheme between Qemu and the systemc component through the remote port or any other interface like LMAC. Do you have any example for it ? is there any solution for that in cosim simulation ? 
Best Regards,  
Ped

hoangvm97, 2023-11-21:
Hi all, In qemu/hw/riscv/virt.c the Cosim is mapped to 0x28000000 but when I dump dtb by qemu-system-riscv64, the cosim isn't in the dtb file, anyone can access to Cosim region?