Xilinx/qemu #44: ERROR: glib-2.48 gthread-2.0 is required to compile QEMU #40

abitofmaya, 2020-07-09:
I have installed most of the dependencies below. I am still getting this error.
```
sudo apt install libglib2.0-dev libgcrypt20-dev zlib1g-dev autoconf automake libtool bison flex libpixman-1-dev libnfs-dev libiscsi-dev libaio-dev libbluetooth-dev libbrlapi-dev libbz2-dev libcap-dev libcap-ng-dev libcurl4-gnutls-dev libgtk-3-dev

../configure --target-list="aarch64-softmmu,microblazeel-softmmu" --enable-debug --enable-fdt --disable-kvm --disable-xen
```

edgarigl, 2020-08-02:
Hi,

Can you show us the exact error you're getting?

wide0s, 2020-10-29:
I'm getting the same issue when compiling QEMU on Ubuntu 20.04 (Focal Fossa). The exact error message is the following:

```
cd /home/wide0s/qemu && ./configure --target-list=arm-softmmu,x86_64-softmmu && make -j2

ERROR: glib-2.48 gthread-2.0 is required to compile QEMU

make: *** [Makefile:36: qemu] Error 1
```
I realized that the problem is missing glib-2.48 and gthread-2.0, so that I installed glib with this command
```
$ apt install libglib2.0-dev
```
It installed version of libglib v2.64:
```
$ dpkg -l | grep libglib
ii  libglib2.0-0:amd64               2.64.2-1~fakesync1                amd64        GLib library of C routines
ii  libglib2.0-bin                   2.64.2-1~fakesync1                amd64        Programs for the GLib library
ii  libglib2.0-data                  2.64.2-1~fakesync1                all          Common files for GLib library
ii  libglib2.0-dev:amd64             2.64.2-1~fakesync1                amd64        Development files for the GLib library
ii  libglib2.0-dev-bin               2.64.2-1~fakesync1                amd64        Development utilities for the GLib library
```
However there is no libgthread2.0-dev or similar which I can install on Ubuntu 20.04 to meet this ./configure script check:
```
glib_req_ver=2.48
glib_modules=gthread-2.0
if test "$modules" = yes; then
    glib_modules="$glib_modules gmodule-export-2.0"
fi
if test "$plugins" = yes; then
    glib_modules="$glib_modules gmodule-2.0"
fi

# This workaround is required due to a bug in pkg-config file for glib as it
# doesn't define GLIB_STATIC_COMPILATION for pkg-config --static

if test "$static" = yes && test "$mingw32" = yes; then
    QEMU_CFLAGS="-DGLIB_STATIC_COMPILATION $QEMU_CFLAGS"
fi

for i in $glib_modules; do
    if $pkg_config --atleast-version=$glib_req_ver $i; then
        glib_cflags=$($pkg_config --cflags $i)
        glib_libs=$($pkg_config --libs $i)
        QEMU_CFLAGS="$glib_cflags $QEMU_CFLAGS"
        LIBS="$glib_libs $LIBS"
        libs_qga="$glib_libs $libs_qga"
    else
        error_exit "glib-$glib_req_ver $i is required to compile QEMU"
    fi
done
```
It'd be great, if someone can have a quick look on this and suggest a patch:)

perror, 2023-02-19:
I stumbled too on this problem (Debian unstable) but I found a workaround (that explain the problem, by the way).

So,  lets start from the beginning. Trying to build QEMU end up with:
```
ERROR: glib-2.48 gthread-2.0 is required to compile QEMU
```
I checked that `gthread-2.0` was present on my distribution which was the case (install `libglib2.0-dev` if it is not installed).

Looking at the code of the `configure` leads to the following test which is failing:
```
#> pkg-config --atleast-version=2.48 gthread-2.0; echo $?
1
```
But, it seems that the pkg-config file of `gthread-2.0` is installed:
```
#> locate  gthread-2.0.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/gthread-2.0.pc
```
So, what is missing? Well, the environment variable `PKG_CONFIG_PATH` can be use to force the distribution to point to the path and find the files:
```
#> export PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig/
#> pkg-config --atleast-version=2.48 gthread-2.0; echo $?
1
```
After that, this should work.

To summarize, this is definitely not a bug in the build system of QEMU. I really think that this should be reported to the maintainer of the `pkg-config` package in order to add `/usr/lib/x86_64-linux-gnu/pkgconfig/` among the list of the default paths for `pkn-config`.