Xilinx/qemu #45: Error building qemu on Ubuntu

BottCode, 2020-07-28:
Hi,

I'm following [these instructions](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842060/QEMU#QEMU-BuildingQEMUSourceCodeforLinux) to build Qemu.
The `./configure` script ends succesfully, while the build process fails. 
Here's the errors.

    make[1]: Entering directory '/home/ubuntu/UNI/qemu/slirp'
    make[1]: Nothing to be done for 'all'.
    make[1]: Leaving directory '/home/ubuntu/UNI/qemu/slirp'
        CHK version_gen.h
      CC      aarch64-softmmu/target/arm/translate-a64.o
    /home/ubuntu/UNI/qemu/target/arm/translate-a64.c: In function ‘disas_ldst’:
    /home/ubuntu/UNI/qemu/target/arm/translate-a64.c:3193:5: error: ‘fn’ may be used uninitialized in this function [- 
    Werror=maybe-uninitialized]
         fn(cpu_reg(s, rt), clean_addr, tcg_rs, get_mem_index(s),
         ^
    /home/ubuntu/UNI/qemu/target/arm/translate-a64.c:3119:22: note: ‘fn’ was declared here
         AtomicThreeOpFn *fn;
                          ^
    cc1: all warnings being treated as errors
    /home/ubuntu/UNI/qemu/rules.mak:69: recipe for target 'target/arm/translate-a64.o' failed
    make[1]: *** [target/arm/translate-a64.o] Error 1
    Makefile:527: recipe for target 'aarch64-softmmu/all' failed
    make: *** [aarch64-softmmu/all] Error 2

How can I solve? I've also tryed to remove the `-Werror` flag from `./dtc/Makefile`, but the result is the same.

edgarigl, 2020-07-29:
Hi, try disable Werror by passing -disable-werror to the configure script.

Best regards,
Edgar

BottCode, 2020-07-29:
Hi @edgarigl, thanks for your answer.
Passing the `--disable-werror` solve that error, but now some others are came up.

    make[1]: Entering directory '/home/ubuntu/UNI/qemu/slirp'
    make[1]: Nothing to be done for 'all'.
    make[1]: Leaving directory '/home/ubuntu/UNI/qemu/slirp'
      CC      block.o
      LINK    qemu-nbd
      LINK    qemu-storage-daemon
      LINK    qemu-img
      LINK    qemu-io
      CC      hw/sd/milkymist-memcard.o
    /home/ubuntu/UNI/qemu/hw/sd/milkymist-memcard.c: In function ‘milkymist_memcard_reset’:
    /home/ubuntu/UNI/qemu/hw/sd/milkymist-memcard.c:257:5: warning: implicit declaration of function ‘device_reset’ [-Wimplicit-      function-declaration]
     device_reset(DEVICE(s->card));
     ^
    /home/ubuntu/UNI/qemu/hw/sd/milkymist-memcard.c:257:5: warning: nested extern declaration of ‘device_reset’ [-Wnested-externs]
    In file included from /home/ubuntu/UNI/qemu/include/hw/qdev-core.h:6:0,
                 from /home/ubuntu/UNI/qemu/include/hw/sysbus.h:6,
                 from /home/ubuntu/UNI/qemu/hw/sd/milkymist-memcard.c:27:
    /home/ubuntu/UNI/qemu/hw/sd/milkymist-memcard.c:257:26: error: ‘MilkymistMemcardState’ has no member named ‘card’
     device_reset(DEVICE(s->card));
                          ^
    /home/ubuntu/UNI/qemu/include/qom/object.h:511:17: note: in definition of macro ‘OBJECT’
     ((Object *)(obj))
                 ^
    /home/ubuntu/UNI/qemu/include/hw/qdev-core.h:15:21: note: in expansion of macro ‘OBJECT_CHECK’
    #define DEVICE(obj) OBJECT_CHECK(DeviceState, (obj), TYPE_DEVICE)
                     ^
    /home/ubuntu/UNI/qemu/hw/sd/milkymist-memcard.c:257:18: note: in expansion of macro ‘DEVICE’
     device_reset(DEVICE(s->card));
                  ^
    /home/ubuntu/UNI/qemu/rules.mak:69: recipe for target 'hw/sd/milkymist-memcard.o' failed
    make: *** [hw/sd/milkymist-memcard.o] Error 1

BottCode, 2020-07-29:
Now it works, thanks.

closed 2020-07-29 by BottCode