Xilinx/qemu #48: Build error on macOS: no member named 'sem' in 'struct QemuSemaphore'

cdelledonne, 2020-09-11:
### Problem

When trying to build QEMU on macOS, I get the following error, in addition to #47:
```
[...]/qemu/hw/core/remote-port.c:496:47: error: no member named 'sem' in 'struct QemuSemaphore'
                sem_getvalue(&s->rx_queue.sem.sem, &sval);
                              ~~~~~~~~~~~~~~~ ^
```

which originates from the definition of `QemuSemaphore` that does not include `sem` when `CONFIG_SEM_TIMEDWAIT` is not defined:
https://github.com/Xilinx/qemu/blob/1eb7d50646fb8f481893135986ca8028c1f6e732/include/qemu/thread-posix.h#L27-L36

### To Reproduce

1. clone the repository
2. comment out `#include <byteswap.h>` in hw/core/remote-port-proto.c to temporarily get rid of #47
3. run
    ```
    mkdir build
    cd build
    ../configure --target-list=arm-softmmu --enable-fdt --disable-xen
    make
    ```

### Environment

- OS: macOS 10.15.6
- Darwin kernel version: 19.6.0
- Compiler: Apple clang version 11.0.0

edgarigl, 2020-09-11:
Hi,

I don't have easy access to a macOS system unfortunately. Any chance you could create a fix and submit a pull-request?

Thanks,
Edgar

cdelledonne, 2020-09-14:
Ok, I'll try to submit a PR soon.

cdelledonne, 2020-09-14:
I just submitted #49 (that should fix #47) and #50 (that should fix this one)

cdelledonne, 2020-09-14:
Closed with 2736c2b1a676fb3f6af87e41ec18e2ba5605ef56

closed 2020-09-14 by cdelledonne