Xilinx/qemu #51: network-buffering

ashwani29, 2020-10-01:
How can I separately run networking buffering thing for qemu while not providing micro-checkpointing tolerance support to it?  I just want to have network buffering support which I can call manually.
Where is the script which do that?

edgarigl, 2020-10-01:
Hi,

I'm afraid I don't quite understand the question. What is the "network buffering thing" you are referring to?
Micro-checkpointing?

Can you elaborate more?

Best regards,
Edgar

ashwani29, 2020-10-01:
Yes I'm talking about micro-checkpointing, in micro-checkpointing, for a duration of the epoch, whatever packets generated inside the VM are buffered. 
I do not want to do VM checkpoints instead periodically call buffering/releasing of the network-output packets.

ashwani29, 2020-10-01:
I hope this clear now?, please reply if it doesn't.

edgarigl, 2020-10-02:
OK thanks for clarifying.

There's no such script available that I'm aware of, so you're likely going to have to develop at least parts of it.
There may be some facilities in QEMU already, but I have a few questions:
What SoC family are you targeting? (Zynq, ZynqMP, Versal, MicroBlaze)
And what net backend are you using? (user/slirp, tap, etc)

Bes regards,
Edgar

ashwani29, 2020-10-02:
what is a Soc? I think its something related to hardware. My machine is intel i5 core. 

I think I've created some confusion here. If you go to this link: https://wiki.qemu.org/Features/MicroCheckpointing  , they have mentioned about how network packets are buffered using `qdiscs` `plug` and `unplug` operations during an epoch. 
https://docs.huihoo.com/doxygen/linux/kernel/3.7/sch__plug_8c_source.html This link shows some more details on the operations performed.  I tried this also but i'm unable to generate a small working example that can simulate buffering and releasing of packets in the host kernel.
The problem is how should I get hold of parameters passed to the functions such as              `struct sk_buff *skb `, `struct nlattr *opt` etc. I already spent two weeks on this and now i'm looking for qemu order of doing it and off course some code script that can help me more so that i can built a custom script which can successfully do this.

edgarigl, 2020-10-02:
Hi,

Note that you have posted an issue on the Xilinx fork of QEMU. Here we deal with things that specifically have to do with the Xilinx SoCs. I think you should post your questions in the upstream QEMU forums.

Best regards,
Edgar

ashwani29, 2020-10-02:
I can't even open an issue on qemu micro-checkpointing GitHub page. what others qemu upstream forums you know? I'll post there.

edgarigl, 2020-10-02:
These are the upstream mailing lists:
https://savannah.nongnu.org/mail/?group=qemu

You could try qemu-discuss or qemu-devel I guess.