Xilinx/qemu #52: Sockets for Cosimulation with SystemC

nvgubba, 2020-10-13:
Hi,
I have a usecase, where i want to run Xilinx QEMU with machine option **virt-cosim** to create TCP socket. Current implementation of Xilinx Qemu is supporting only **Unix Socket** for remote port Cosimulation. Is there a way to create a TCP Socket?

I want to run QEMU on one machine and the HW model (systemC) to run on another machine.  I tried to change the **rp_autocreate_chardesc** function in **remote-port.c** file, but it doesn't seems to work

Following is the change, i changed the string pattern to **tcp** from **unix** of the following line

`    r = asprintf(&chardesc, "tcp:%s/qemu-rport-%s%s",
                   machine_path, prefix, server ? ",wait,server" : "");
`

Can you give some suggestions how to fix the problem ?

edgarigl, 2020-10-13:
Hi,

Here are some examples on how to use TCP sockets. You'll have to replace the 192.168.10.100 with the ip-address of your peer.

On QEMU's command-line add the following:
-chardev socket,id=cosim,host=192.168.10.100,port=5555

On the SystemC programs command-line, replace the unix:// socket description with the following:
tcpd://0.0.0.0:5555

With these options, you need to start the SystemC program first, it'll create a listening TCP socket and QEMU will connect to it.
The 0.0.0.0 IP address to the SystemC program means that it'll listen on all IP addresses configured on that host. You can also replace the 0.0.0.0 with a specific address such as 192.168.10.100 in this example.

Best regards,
Edgar

nvgubba, 2020-10-13:
Hi Edgar,
Thanks for your reply. I am using the command like this
**`sudo  /home/open-silicon/work/xlinxqemu/inst/bin/qemu-system-riscv64 -M virt-cosim -smp 4 -serial stdio -display none -m 2G -dtb virt.dtb -bios /home/open-silicon/work/xlinxqemu/pc-bios/opensbi-riscv64-virt-fw_jump.bin -kernel riscv-pk/build/bbl -append "root=/dev/vda ro console=ttyS0" -drive file=busybear-linux/busybear.bin,format=raw,id=hd0 -device virtio-blk-device,drive=hd0 -object rng-random,filename=/dev/urandom,id=rng0 -device virtio-rng-device,rng=rng0 -device virtio-net-device,netdev=usernet -netdev user,id=usernet -netdev user,id=net4 -device remote-port-net,rp-adaptor0=/machine/cosim,rp-chan0=256,rp-chan1=266,netdev=net4 -chardev socket,id=cosim,host=127.0.0.1,port=5555
`**
But i am getting error as below, could you please let me know..what arguments need to be corrected

**`qemu-system-riscv64: -device remote-port-net,rp-adaptor0=/machine/cosim,rp-chan0=256,rp-chan1=266,netdev=net4: Did not find rp adaptor /machine/cosim!`**

Thanks and Regards
Rajesh

edgarigl, 2020-10-13:
Hi,

Hmm, strange, I can't spot it. Do you have any changes in QEMU?

Here's my full command-line using QEMU from our master branch (no patches):
```
qemu-system-riscv64  -M virt-cosim  -smp 4 -serial stdio -display none -m 2G -dtb images/riscv/posh-lmac/virt.dtb -kernel images/riscv/posh-lmac/Image -append "root=/dev/vda ro console=ttyS0" -drive file=images/riscv/posh-lmac/rootfs.ext2,format=raw,snapshot=on,id=hd0 -device virtio-blk-device,drive=hd0 -object rng-random,filename=/dev/urandom,id=rng0 -device virtio-rng-device,rng=rng0 -device virtio-net-device,netdev=usernet -netdev user,id=usernet -netdev user,id=net4 -device remote-port-net,rp-adaptor0=/machine/cosim,rp-chan0=256,rp-chan1=266,netdev=net4  -machine-path /tmp/machine-riscv/ -chardev socket,id=cosim,host=127.0.0.1,port=5555
```

nvgubba, 2020-10-13:
Hi,

Sorry its my mistake..i thought **machine-path** is not required when its a tcp socket. Now it is working after adding **-machine-path**

edgarigl, 2020-10-13:
No worries, I'm happy it works now :-)