Xilinx/qemu #53: Creating and reloading snapshots for zynqmp with cosim

asultan123, 2020-11-12:
I'm trying to create and load snapshots for the Qemu side using the qemu monitor. With a separate qemu instance for the zynqmp PMU I can't seem to load a saved VM regardless of how I save a snapshot (live save or stop then save). If I try to save and load two separate a snapshot for the the zynqmp and pmu it still doesn't work. By doesn't load I mean just hangs on startup if I run the following command with -loadvm TAG. 

`$QEMU_PATH/aarch64-softmmu/qemu-system-aarch64 -M arm-generic-fdt   -serial mon:stdio -serial /dev/null -display none -device loader,file=$IMAGE_PATH/images/linux/bl31.elf,cpu-num=0 -device loader,file=$IMAGE_PATH/images/linux/Image_Modified,addr=0x00080000 -device loader,file=$IMAGE_PATH/images/linux/system_Modified.dtb,addr=0x1407f000 -device loader,file=$IMAGE_PATH/images/linux/linux-boot_Modified.elf -gdb tcp::9000 -dtb $IMAGE_PATH/images/linux/system_Modified.dtb  -net nic -net nic -net nic -hw-dtb $IMAGE_PATH/images/linux/zcu102-arm.cosim.dtb -machine-path /tmp/tmp.xilinxcosim -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true -drive if=none,format=qcow2,file=dummy1.qcow2 -net user,hostfwd=tcp::2222-:22 -net nic -m 4G -loadvm TAG
`  

I'm unsure if snapshotting is supported or not in this case or if I'm doing something wrong. I followed this tutorial: https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/28737647/QEMU+LibSystemCTLM-SOC and used a different image built with petalinux. 
I'm a little inexperienced with using qemu so I apologize if part of my question don't make a lot of sense.

asultan123, 2020-12-11:
I can't say I solved this one but since the issue is still open here are my findings. 
Regardless of how you snapshot via QEMU it hangs on loading a snapshot. I tried using CRIU to save the process of the QEMU Instance for both the microblaze and zynqmp instances + the SystemC side and that sort of works but for the particular kernel I was running it crashes on process reload. Unfortunately, my understanding of both QEMU and CRIU's internals isn't sophisticated enough to figure out why the kernel running in the QEMU instance crashes.The only concrete snapshotting mechanism I could get working was if I ran the qemu instance in a virtual machine but obviously that's a little bit naive. Reloading a snapshot in virtualbox takes as long as firing up a new QEMU instance so there's that. I'll eventually get back to this if I find something that makes snapshotting work. Till then, I hope this project isn't dead and someone from the dev team gets back to it.

ho28, 2023-10-13:
Hi, I stumbled across this reported issue after finding out that creating and loading snapshots for simulated Versal on arm-generic-fdt machine also encounters the same problem.

I discovered that this is due to the way memory is registered during the arm-generic-fdt machine's memory initialization routine (init_memory) that results in the ram not being registered with the vmstate save handler.

I believe this patch fixes the issue, please let me know if I should file a pull request.

[0001-arm-generic-fdt-Save-RAM-during-snapshot.patch](https://github.com/Xilinx/qemu/files/12888774/0001-arm-generic-fdt-Save-RAM-during-snapshot.patch)