Xilinx/qemu #54: Remote Port Implementation for Multi-Arch QEMU

balvansh, 2021-03-02:
Hello,

Can you please point me to the documentation that highlights the implementation and the internal workings of the remote port which is used to connect the ZynqMP Ultrascale+ MPSoC QEMU instance to its MicroBlaze PMU QEMU instance? Also, I want to establish a connection between two APUs using the remote port, and would like to know if this is possible using the current implementation of remote ports?

Thank you

edgarigl, 2021-03-02:
Hi,

The remote-port documentation is mostly in source-code. You can find it here:

SystemC:
https://github.com/Xilinx/libsystemctlm-soc/blob/master/libremote-port/
https://github.com/Xilinx/libsystemctlm-soc/blob/master/libremote-port/README
https://github.com/Xilinx/libsystemctlm-soc/blob/master/libremote-port/remote-port-proto.h

The QEMU port is mainly used by the Xilinx machines that get instantiated with device-tree files.
An example of the Versal connections can be found here:
https://github.com/Xilinx/qemu-devicetrees/blob/master/versal-ps-pl-remoteport.dtsi

If you're looking to connect machines that are builtin to QEMU (C coded machines), we've got an example for RISCV:
https://github.com/Xilinx/qemu/blob/master/hw/riscv/virt.c#L494

When connecting QEMU to QEMU (and some times also to SystemC) it's useful to use Remote-Port for register accesses and shared memory files for RAM sharing between processes. Ram sharing speeds things up alot but it only works if machines are on the same host. On the Xilinx multi-arch setups, we always use RAM sharing between the ARM and MicroBlaze QEMU instances.

It would be good if you could clarify a little more what you're trying to do, we may be able to give better help.

Good luck and Best regards,
Edgar

balvansh, 2021-03-02:
Hello Edgar,

Thank you for the reply. For my use case, I am trying to connect the QEMU Zynq7000 device to a MicroBlaze device (not as PMU, but instead as a general device (I understand to bring up such configuration I will have to generate a custom device-tree file)) and also other QEMU built-in devices to each other using remote-port.

When you say there is RAM being shared among the two connected QEMU instances, how much amount of RAM is actually shared between the two instances? And is there a way to increase or decrease the amount of RAM shared? 

And are there any examples to understand how remote-port can be used to connect two QEMU instances for my use case?

Thank you