Xilinx/qemu #55: arrow keys is not working in petalinux when use XilinxQemu from windows sdk

minamagdykamal, 2021-03-02:
Hello All,

arrow Keys is not working in xilinx qemu prebuilt image provided in Xilinx SDK v2018.3 window10 platform using prebuilt [petalinux ](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html)provided by xilinx.

steps:

- run pmu =>qemu-system-microblazeel -M microblaze-fdt -nographic -display none -kernel pre-built/linux/images/pmu_rom_qemu_sha3.elf -device loader,file=images/linux/pmufw.elf -hw-dtb images/linux/zynqmp-qemu-multiarch-pmu.dtb -machine-path ./qemu-memory -device loader,addr=0xfd1a0074,data=0x1011003,data-len=4 -device loader,addr=0xfd1a007C,data=0x1010f03,data-len=4 -global remote-port.chardesc=tcp:127.0.0.1:9000,server

- run APU,RPU => qemu-system-aarch64 -M arm-generic-fdt -nographic -k en-us -serial mon:stdio -serial telnet::1235,server,nowait -device loader,file=images/linux/bl31.elf,cpu-num=0 -device loader,file=images/linux/Image,addr=0x00080000 -device loader,file=images/linux/system.dtb,addr=0x1407f000 -device loader,file=build/misc/linux-boot/linux-boot.elf -gdb tcp::1234 -dtb images/linux/system.dtb -hw-dtb images/linux/zynqmp-qemu-multiarch-arm.dtb -machine-path ./qemu-memory -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true -m 2G -drive file=qemu-sd.img,if=sd,format=raw,index=1 -monitor telnet:127.0.0.1:55555,server,nowait -global remote-port.chardesc=tcp:127.0.0.1:9000 -net nic -net nic -net nic -net nic,netdev=eth0 -netdev user,id=eth0,net=192.168.144.0/24,dhcpstart=192.168.144.11,hostfwd=tcp::21-:21,hostfwd=tcp::22-:22,hostfwd=tcp::1534-:1534,hostfwd=udp::14540-:14540,hostfwd=udp::14541-:14541,hostfwd=tcp::14542-:14542,hostfwd=tcp::14543-:14543,tftp=shared

linux booted successfully and everything works ok, except the arrow keys

edgarigl, 2021-03-02:
Hi,

I'm sorry but I wouldn't know what's wrong. You might want to try running QEMU in a different shell/terminal?
In what shell are you running the commands now?

Thanks,
Edgar

minamagdykamal, 2021-03-02:
Hello Edgar,

many thanks for your quick respond and for your support.

i had tried it with different terminals:

- WSL terminal.
- MobaXterm.
- Xilinx Software Command Line Tool (xsct).
- Windows Command Prompt.
- PowerShell.
- Bash.
- cygwin.

 but still the same issue,

closed 2021-03-24 by minamagdykamal