Xilinx/qemu #57: Booting qnx os with qemu

asifsid-32, 2021-04-27:
Hello,

I am trying to boot QNX OS on top of qemu using QNX IFS (aarch64). 

Following command is being used - 

qemu-system-aarch64 -machine xlnx-zcu102 -m 1G -kernel IFS.bin -drive file=filesystem.qcow2,if=none,id=drv0 -nographic -serial mon:stdio 

Using this it gives a shutdown error and the kernel does not load.

Can someone help how we can boot the QNX kernel on top of qemu with xilinx machine ? 

Thanks

edgarigl, 2021-04-28:
Hi,

A couple of questions:

What do you have in the IFS.bin binary? Is there only the QNX OS expected to run at EL1? Or is there SW that expects to run at EL3? The command-line will depend on that.

Are you able to run this on real HW? Are you in that case running it together with ARM Trusted Firmware and the PMU FW from Xilinx? or are you using something else?

Best regards,
Edgar

asifsid-32, 2021-04-29:
Hii @edgarigl 

Thanks for the comment.
We were able to boot the QNX OS on QEMU with xilinx machine using the following command - 

./qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -display none -device loader,file=QNX-IFS.bin,cpu-num=0 -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true

Regards
Asif

edgarigl, 2021-04-29:
Hi!

I suggest that you don't use the xlnx-zcu102 machine but our more complete device-tree based machine, those are a lot more complete.
You're going to need to build the device-trees from here:
https://github.com/Xilinx/qemu-devicetrees

There's more information here:
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/821395464/QEMU+User+Documentation

To run something like a binary from EL3, here's an example command-line:
qemu-system-aarch64 -M arm-generic-fdt  -serial stdio -display none  -m 2G  -hw-dtb dts/LATEST/SINGLE_ARCH/zcu102-arm.dtb -device loader,file=QNX-IFS.bin,cpu-num=0 -device loader,addr=0xfd1a0104,data=0x8000000e,data-len=4

If you need to run with ATF and perhaps the PMU FW, I can provide other examples but it would be good to get more details about your QNX-IFS.bin, i.e what address does it expect to start from? address 0? What EL level?

Best regards,
Edgar

asifsid-32, 2021-04-30:
Hii @edgarigl 

Thanks for the inputs. 
We are trying to configure/generate IPI interrupts for which we have created a separate issue ([Link-](https://github.com/Xilinx/qemu/issues/58)). Here we are trying to run everything in the same core. 

Can you please suggest something on how we can configure the interrupt ? 

Regards
Asif

edgarigl, 2021-05-02:
The interrupt you mentioned is not yet connected in the xlnx-zcu102 model, we'll have to fix that in upstream. In the meantime you'll need to use the device-tree machines.

asifsid-32, 2021-05-02:
> The interrupt you mentioned is not yet connected in the xlnx-zcu102 model, we'll have to fix that in upstream. In the meantime you'll need to use the device-tree machines.

Hii @edgarigl 

Thanks for the reply !

1) Can you please help us with the steps we need to follow for using device-tree machines for configuring interrupts ? 

2) IPI interrupts can be configured/generated using device-tree machines ? If yes, how can we do IPI interrupts or any other interrupts ? 

3) When can we expect the IPI in upstream ? 

Regards
Asif

asifsid-32, 2021-05-03:
> Hi!
>
> I suggest that you don't use the xlnx-zcu102 machine but our more complete device-tree based machine, those are a lot more complete.
> You're going to need to build the device-trees from here: [...]

Hello Edgar,

Please find attached the minimal-build file used for generating the QNX-IFS. Address related info can be checked in build file. 
EL level i am not sure as i am new to this. 

I tried the command suggested by you - 

qemu-system-aarch64 -M arm-generic-fdt -serial stdio -display none -m 2G -hw-dtb dts/LATEST/SINGLE_ARCH/zcu102-arm.dtb -device loader,file=QNX-IFS.bin,cpu-num=0 -device loader,addr=0xfd1a0104,data=0x8000000e,data-len=4

but it is not booting the kernel and just hangs after executing. Any other suggestions for interrupts configuration ? 
[minimal-build.txt](https://github.com/Xilinx/qemu/files/6415403/minimal-build.txt)

Thanks
Asif

edgarigl, 2021-05-04:
Hi,

Can you try adding -machine linux=yes, e.g:

qemu-system-aarch64 -M arm-generic-fdt -serial stdio -display none -m 2G -hw-dtb dts/LATEST/SINGLE_ARCH/zcu102-arm.dtb -device loader,file=QNX-IFS.bin,cpu-num=0 -device loader,addr=0xfd1a0104,data=0x8000000e,data-len=4 -machine linux=yes

Best regards,
Edgar

asifsid-32, 2021-05-08:
Hii @edgarigl 

I tried the above command with -machine linux=yes, but it is not booting the kernel and just hangs after executing. Any other suggestions for interrupts configuration ?

Regards 
Asif

aravie, 2021-08-05:
@asifsid-32 Can you share me the procedure to use qemu to bringup ZCU-102 with QNX kernel image. Thank you very much in advance.

XVilka, 2022-09-13:
@asifsid-32 have you had any luck with this?

charantejaabbavathini, 2024-02-06:
Hi @edgarigl, @asifsid-32 
I am trying to boot qnx os using below cmd

qemu-system-aarch64 -M xlnx-zcu102 -m 2G -serial mon:stdio -display none -device loader,file=BOOT.BIN,cpu-num=0 -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true

Here is the output I got

~/qnx710/output$ qemu-system-aarch64 -M xlnx-zcu102 -m 2G -serial mon:stdio -display none -device loader,file=BOOT.BIN,cpu-num=0 -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true
pulseaudio: set_sink_input_volume() failed
pulseaudio: Reason: Invalid argument
pulseaudio: set_sink_input_mute() failed
pulseaudio: Reason: Invalid argument
qemu-system-aarch64: warning: global xlnx,zynqmp-boot.cpu-num has invalid class name
qemu-system-aarch64: warning: global xlnx,zynqmp-boot.use-pmufw has invalid class name
QEMU: Terminated

I see Qemu opens up but it did not boot the OS. Any leads would be helpful!

Thanks!

charantejaabbavathini, 2024-02-07:
Hi @edgarigl @asifsid-32,

I have tried to boot the qnx os aarch64 using 
qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -display none -device loader,file=QNX-IFS.bin,cpu-num=0 -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true

Here is the output of the above command:
~/qnx710/output$ qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -display none -device loader,file=QNX-IFS.bin,cpu-num=0 -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true
pulseaudio: set_sink_input_volume() failed
pulseaudio: Reason: Invalid argument
pulseaudio: set_sink_input_mute() failed
pulseaudio: Reason: Invalid argument
qemu-system-aarch64: warning: global xlnx,zynqmp-boot.cpu-num has invalid class name
qemu-system-aarch64: warning: global xlnx,zynqmp-boot.use-pmufw has invalid class name
QEMU: Terminated

I think, the qemu is up but its unable to boot the OS. The QNX-IFS.bin file is only qnx image file. Am I missing something to pass like the binary file boot.bin or should i include the boot.bin in the QNX-IFS.bin? Any leads would be helpful.

Thank you!

Prem-48085, 2024-06-17:
Is this problem solved ? How can I get all the necessary file required to boot ARM QNX on Qemu. Please help me if you have any sample image etc to boot.

rmadhu90, 2025-05-12:
> Hii [@edgarigl](https://github.com/edgarigl)
> 
> I tried the above command with -machine linux=yes, but it is not booting the kernel and just hangs after executing. Any other suggestions for interrupts configuration ?
> 
> Regards Asif

@asifsid-32 , please let me know whether you could resolve this issue