Xilinx/qemu #59: Enabling Software Generated Interrupts

asifsid-32, 2021-05-05:
Hello All,

We are trying to enable/generate software generated interrupts in ZCU102. We are using following sequence to write to the GIC Registers -

       Read(0xF9020000)  -- Reading GICC_CTLR gives the value as    0x00000001
       Read(0xF9010004 )  -- Reading GICD_TYPER gives the value as    0x00000065 

-       Write (0xF9010000 , 0x00000003 )                      // GICD_CTLR Register
- 	Read (0xF9010000 )  
- 	Write (0xF9010084 , 0x00000001)                       // GICD_IGROUPR1 Register
- 	Read (0xF9010084 )
- 	Read(0xF9010100 )                                            // GICD_ISENABLER0  Register
- 	Write(0xF9010F00 , 0x00010000)                      //GICD_SGIR Register
- 	Read(0xF9010F00 )       

Commands we used writing and reading to the registers- 

- out32 0xF9010000 0x00000003     // writing  value - 0x00000003 to GICD_CTLR register with address 0xF9010000
- in32 0xF9010000                            // Reading value from address  0xF9010000
- out32 0xF9010084 0x00000001
- in32 0xF9010084
- in32 0xF9010100
- out32 0xF9010F00 0x00010000
- in32 0xF9010F00

But we are not able to set the SGIR Register which gives "0" when read after writing in the above sequence. Could you please suggest the possible reason why we are not able to set the SGIR Register or something we are missing in the sequence.

Regards 
Asif

franciscoIglesias, 2021-05-12:
Hello Asif,

For generating SGIs I think it would be best to try to follow the GICv2 specification [1] (there will be several configuration steps involved, and also security state affects register reads writes if security extensions are enabled).

To note here is that the GICv2 in the ZynqMP implements security extensions so when I read out GICD_TYPER I get 0x0465 (I'm launching QEMU with the zcu102 device tree).

About above procedure, since I don't see any SGI priority configuration in the sequence that would be my best guess at the moment to why no SGI is being generated. Also, GICD_SGIR is a write only register [1] which is why you are always reading 0 (QEMU returns 0). 

Best regards,
Francisco Iglesias

[1] ARM Generic Interrupt Controller Architecture version 2.0 - Architecture Specification

asifsid-32, 2021-05-13:
Hi Francisco

We tried writing GICD_IGROUPR0 register for secure mode but the GICD_TYPER is reading 0x00000065 which means that the secure bit is not getting enabled. We have also tried setting the priority registers but the result was same.

Could you please suggest the steps required to implement the GIC in secure mode and as to enable software interrupt SGIR.

Regards 
Asif

asifsid-32, 2021-05-18:
Hello,

In addition to above, we also tried following sequence for triggering software interrupt - 
 
1. map to CPU by writing the following register
GICD_ITARGETSRn to 0x00000001

2. Interrupt Set-Enable by Writing 1 to a Set-enable bit enables forwarding of the interrupt
GICD_ISENABLERn to 0x00000001

3. Enables the forwarding of pending interrupts by writing 1 
GICD_CTLR to 0x00000001

4. Trigger the SW interrupt by writing appropriate values (SW channel number ) into GICD_SGIR register
GICD_SGIR to 0x00010000

5. Reading pending bits register values
a. GICD_SPENDSGIRn
b. GICD_ISPENDRn

Observation - 
 We are not able to set the SGIR Register which gives "0" when read after writing in the above sequence. 
 
 Could you please suggest the possible reason why we are not able to set the SGIR Register or something we are missing in the sequence.
 
 Regards 
 Asif