Xilinx/qemu #61: Netowrking in QEMU with xilinx-zcu102 machine

asifsid32, 2021-06-08:
Hello All, 

I am working on the networking between my host OS linux and guest OS QNX(Running on top of qemu). I have an interface xzynq0 in my guest OS with a static/dynamic IP address. 

I am executing the qemu command as follows - 

qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -display none -device loader,file=QNX_IFS.bin,cpu-num=0 -netdev bridge,br=virbr0,id=net0,helper=/usr/lib/qemu/qemu-bridge-helper -nographic  -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true
 
But the observation is when i try to ping the guest OS with some IP from my host OS, it is showing destination unreachable or when i try to ping from my guest to host it shows host is down. 

I do have a virbr0 bridge in my host os - 

virbr0    Link encap:Ethernet  HWaddr 00:00:00:00:00:00  
          inet addr:192.168.122.1  Bcast:192.168.122.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B) 

And in my guest OS the interface is - 

xzynq0: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> mtu 1500
address: 00:0a:35:02:78:10
media: Ethernet 100baseTX full-duplex
status: active
inet 192.168.1.2 netmask 0xffffff00 broadcast 192.168.1.255
inet6 fe80::20a:35ff:fe02:7810%xzynq0 prefixlen 64 scopeid 0x11

The same xzynq0 interface works in the actual target hardware. 
Can you please help me out here what am i missing to enable the networking. Is it the case that bridge is not getting set correctly ? Or what can be the possible reasons for the same ? 

Regards
Asif

edgarigl, 2021-06-12:
The ZynqMP SOC has 4 Ethernet MAC's. On the ZCU102 board, GEM3 is the one to use.
You'll need to connect your bridge to the GEM3 model by adding 3 -net nic command-line options.

eg.:

-net nic,model=cadence_gem -net nic,model=cadence_gem -net nic,model=cadence_gem,netdev=net0

Best regards,
Edgar

asifsid-32, 2021-07-01:
> The ZynqMP SOC has 4 Ethernet MAC's. On the ZCU102 board, GEM3 is the one to use.
> You'll need to connect your bridge to the GEM3 model by adding 3 -net nic command-line options.
> 
> eg.:
> 
> -net nic,model=cadence_gem -net nic,model=cadence_gem -net nic,model=cadence_gem,netdev=net0
> 
> Best regards,
> Edgar

Hii Edgar, 

Thanks for the input. The above command options Worked.

I am executing the qemu command as follows - 

qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -display none \
-device loader,file=QNX-IFS.bin,cpu-num=0 \ 
-net nic,model=cadence_gem -net nic,model=cadence_gem -net nic,model=cadence_gem -net nic,model=cadence_gem,netdev=xzynq0 -netdev user,id=xzynq0,tftp=/tftpboot 

After booting the guest OS following is the output of ifconfig - 

# ifconfig
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> mtu 33136
	inet 127.0.0.1 netmask 0xff000000
	inet6 ::1 prefixlen 128
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1
xzynq0: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	capabilities rx=7<IP4CSUM,TCP4CSUM,UDP4CSUM>
	capabilities tx=0
	enabled=0
	address: 00:00:00:00:00:00
	media: Ethernet autoselect (1000baseT full-duplex,flowcontrol)
	status: active
	inet 10.0.2.15 netmask 0xffffff00 broadcast 10.0.2.255
	inet6 fe80::401a:a90e:79f:a326%xzynq0 prefixlen 64 scopeid 0x11

I am able to transfer files successfully between host os linux and guest using SCP when ssh is launched. 

But i am not able to ping from guest to host or host to guest.  Also if i want to debug a qnx application using gdb remote debug it does not work. 
Is it because of QEMU user networking(SLIRP) and it does not support other Network services ?

I tried following the steps given in the "Enabling ping in the guest, on Linux hosts" section from the link https://wiki.qemu.org/Documentation/Networking#User_Networking_.28SLIRP.29 
but it did not work. 

Can you please suggest what changes are required in qemu command line so that i can ping from guest to host or host to guest as this is required for remote debug.

Regards 
Asif

shizhuang-li, 2022-08-11:
Hello,Edgar and Asif, Your discussion is very interesting, I am also trying to build a QNX os on qemu recently, I tried Asif's method, as shown below:
`qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio  -device loader,file=/root/xlinxbsp/images/QNX-IFS,cpu-num=0  -nographic -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true`
The QNX-IFS file is the xlnx-zcu102 BSP package I downloaded from the QNX software center.When I run the above command, nothing is displayed, and the command cannot be terminated, only the kill command can be used to end the process.
[image]

Regards
Shizhuang.li

codectq92, 2022-12-28:
> Hello,Edgar and Asif, Your discussion is very interesting, I am also trying to build a QNX os on qemu recently, I tried Asif's method, as shown below: `qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -device loader,file=/root/xlinxbsp/images/QNX-IFS,cpu-num=0 -nographic -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true` The QNX-IFS file is the xlnx-zcu102 BSP package I downloaded from the QNX software center.When I run the above command, nothing is displayed, and the command cannot be terminated, only the kill command can be used to end the process. [image] I'm guessing that there may be a problem with my QNX-IFS file, but I'm not sure what the specific problem is? So I would like to ask how to solve this problem, and how did you get the QNX-IFS file?
> 
> Regards Shizhuang.li

I have the same question, did you sloved it now?

FaiqueAli, 2024-02-15:
Hi everyone,
I am also running into the same issue did anyone find a solution for this?
when I hit the command to create a VM, the terminal hangs with the following error...

`qemu-system-aarch64: warning: hub 0 is not connected to host network`

looking for further information thanks :)