Xilinx/qemu #62: Timer Precision in QEMU

asifsid-32, 2021-07-30:
Hello All,

We are triggering software generated interrupts in qnx OS running on top of QEMU 5.2.0
While doing the profiling of the Interrupts getting triggered and received we could see that the timings are not matching with the one's configured.
The timer configured for 5ms shows the timing as 10ms consistently. When checked the same in target HW it works correctly and prints the timing as 5ms.

What can be the possible reasons for this issue. Do we need to add additional qemu command line option while booting OS ?

Regards
Asif Siddiqui

edgarigl, 2021-07-30:
Hi,

QEMU has a lower limit on timers at 10us.
Have a look at hw/core/ptimer.c.

But 5ms is well above that, my guess is that you may be seeing a limitation on your host?
What timer device are you programing? The ARM generic timers?

Best regards,
Edgar

asifsid-32, 2021-08-02:
Hi Edgar,

Yes we were seeing a limitation on our host which was then changed with better configuration system. Still we can see a mismatch between the configured timing and the Interrupt triggering timing. 

The Timer being used is QNX specific. Link - http://www.qnx.com/developers/docs/6.5.0/index.jsp?topic=%2Fcom.qnx.doc.neutrino_lib_ref%2Fc%2Fclock_gettime.html

 
Can you please suggest what can be the possible reasons for this issue ? Is there something on QEMU configuration we can add to avoid this system configuration issue
 
Best Regards
Asif