Xilinx/qemu #64: Is it possible to use remote-port with a standard ARM machine?

ninipa, 2021-09-07:
Hi,

Is it possible to only utilize remote-port in xilinx-qemu, but using a standard ARM machine like "virt"?
What I want is just using my own kernel and filesystem, but makes the machine communicate with systemc modules outside.
If so, could you please give some suggestions about how to do it?
Thank you!

ninipa, 2021-09-07:
In addition, I compared ./hw/riscv/virt.c in original QEMU(5.1.0) and xilinx-qemu, looks like it's possible to hard-coded remote-port in a machine.
Can I do similar thing to ./hw/arm/virt.c?

franciscoIglesias, 2021-09-16:
Hi ninipa,

Yes it should be possible to modify an Arm machine similarly as above with the riscv. For studying above riscv remote port connection setup you could try to launch the demo and look into the code (all is on github), instructions are here: https://github.com/Xilinx/systemctlm-cosim-demo/blob/master/docs/lmac-demos.md.

Best regards,
Francisco Iglesias