Xilinx/qemu #66: Compile errors

SJ123SJ, 2021-12-13:
I was following the Confluence build [directions](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/822312999/Building+and+Running+QEMU+from+Source+Code)

I used the latest branches 2020.3 and 2021.1
Both builds resulted in the same errors
```
/home/sj/temp/qemu/block/vpc.c: In function ‘vpc_open’:

/home/sj/temp/qemu/block/vpc.c:358:51: error: array subscript ‘VHDDynDiskHeader {aka struct vhd_dyndisk_header}[0]’ is partly outside array bounds of ‘uint8_t[512]’ {aka ‘unsigned char[512]’} [-Werror=array-bounds]

  358 |         s->block_size = be32_to_cpu(dyndisk_header->block_size);

      |                                                   ^~

/home/sj/temp/qemu/block/vpc.c:223:13: note: while referencing ‘buf’

  223 |     uint8_t buf[HEADER_SIZE];

      |             ^~~

/home/sj/temp/qemu/block/vpc.c:366:58: error: array subscript ‘VHDDynDiskHeader {aka struct vhd_dyndisk_header}[0]’ is partly outside array bounds of ‘uint8_t[512]’ {aka ‘unsigned char[512]’} [-Werror=array-bounds]

  366 |         s->max_table_entries = be32_to_cpu(dyndisk_header->max_table_entries);

      |                                                          ^~

/home/sj/temp/qemu/block/vpc.c:223:13: note: while referencing ‘buf’

  223 |     uint8_t buf[HEADER_SIZE];

      |             ^~~

/home/sj/temp/qemu/block/vpc.c:398:51: error: array subscript ‘VHDDynDiskHeader {aka struct vhd_dyndisk_header}[0]’ is partly outside array bounds of ‘uint8_t[512]’ {aka ‘unsigned char[512]’} [-Werror=array-bounds]

  398 |         s->bat_offset = be64_to_cpu(dyndisk_header->table_offset);

      |                                                   ^~

/home/sj/temp/qemu/block/vpc.c:223:13: note: while referencing ‘buf’

  223 |     uint8_t buf[HEADER_SIZE];

      |             ^~~

  CC      block/qcow2-threads.o

cc1: all warnings being treated as errors

make: *** [/home/sj/temp/qemu/rules.mak:69: block/vpc.o] Error 1

make: *** Waiting for unfinished jobs....

  CC      block/qed.o
```

How to best proceed?

SJ123SJ, 2021-12-14:
I tried replacing vpc.c above with a version found in the QEMM project [here](https://gitlab.com/qemu-project/qemu/-/blob/master/block/vpc.c). The above compilation errors go away, but new errors surface:
```
/home/sj/temp/qemu/block/vpc.c:1238:35: error: initialization of ‘int (*)(BlockDriverState *, uint64_t,  uint64_t,  QEMUIOVector *, int)’ {aka ‘int (*)(BlockDriverState *, long unsigned int,  long unsigned int,  QEMUIOVector *, int)’} from incompatible pointer type ‘int (*)(BlockDriverState *, int64_t,  int64_t,  QEMUIOVector *, BdrvRequestFlags)’ {aka ‘int (*)(BlockDriverState *, long int,  long int,  QEMUIOVector *, BdrvRequestFlags)’} [-Werror=incompatible-pointer-types]

 1238 |     .bdrv_co_preadv             = vpc_co_preadv,

      |                                   ^~~~~~~~~~~~~

/home/sj/temp/qemu/block/vpc.c:1238:35: note: (near initialization for ‘bdrv_vpc.bdrv_co_preadv’)

/home/sj/temp/qemu/block/vpc.c:1239:35: error: initialization of ‘int (*)(BlockDriverState *, uint64_t,  uint64_t,  QEMUIOVector *, int)’ {aka ‘int (*)(BlockDriverState *, long unsigned int,  long unsigned int,  QEMUIOVector *, int)’} from incompatible pointer type ‘int (*)(BlockDriverState *, int64_t,  int64_t,  QEMUIOVector *, BdrvRequestFlags)’ {aka ‘int (*)(BlockDriverState *, long int,  long int,  QEMUIOVector *, BdrvRequestFlags)’} [-Werror=incompatible-pointer-types]

 1239 |     .bdrv_co_pwritev            = vpc_co_pwritev,

      |                                   ^~~~~~~~~~~~~~

/home/sj/temp/qemu/block/vpc.c:1239:35: note: (near initialization for ‘bdrv_vpc.bdrv_co_pwritev’)
```

edgarigl, 2021-12-16:
Hi Jeff,

What compiler and version of it are you using?

When building QEMU from older branches you may need to use an older toolchain or configure with --disable-werror.

Best regards,
Edgar

SJ123SJ, 2021-12-16:
I'm using gcc/g++ 11.2.0
I was following the build instructions [here](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/822312999/Building+and+Running+QEMU+from+Source+Code). The reported errors are from building using the branch listed in the build instructions.  Since the documented branch is "old", I switched to build from the master. Which BTW now requires one to install NINJA, so I did... and now the code compiles.

closed 2022-01-17 by edgarigl