Xilinx/qemu #71: How to run QNX OS on qemu

shizhuang-li, 2022-08-11:
Hello,Edgar and Asif, Your discussion is very interesting, I am also trying to build a QNX os on qemu recently, I tried Asif's method, as shown below:
qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -device loader,file=/root/xlinxbsp/images/QNX-IFS,cpu-num=0 -nographic -global xlnx,zynqmp-boot.cpu-num=0 -global xlnx,zynqmp-boot.use-pmufw=true
The QNX-IFS file is the xlnx-zcu102 BSP package I downloaded from the QNX software center.When I run the above command, nothing is displayed, and the command cannot be terminated, only the kill command can be used to end the process.
aMfATYRpBC
I'm guessing that there may be a problem with my QNX-IFS file, but I'm not sure what the specific problem is? So I would like to ask how to solve this problem, and how did you get the QNX-IFS file?

Regards
Shizhuang.li

tsi-345, 2023-07-17:
Hi Shizhuang.li
you have to add the QNX-IFS startaddress:
qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -device loader,file=/root/xlinxbsp/images/QNX-IFS,addr=0x00100000,cpu-num=0 -nographic -global xlnx,zynqmp-boot.cpu-num=0 -global `xlnx,zynqmp-boot.use-pmufw=true

This works for me.

FaiqueAli, 2024-01-30:
Hi, I found this post helpful for creating the QNX VM, as I am struggling to create VM for QNX using Qemu.
after creating the VM using this way, I found that the VM do not have complete directory structure like /mnt/app folder and some other folder also missing.
while I also tried `-device loader,file=rootfs.cpio.gz.u-boot,addr=0x04000000,force-raw=on ` but the result is same.
I wanted to know that am I missing some setting?

Second, what is the difference between M arm-generic-fdt and xlnx-zcu102  machine type in Qemu?

Thanks :)

charantejaabbavathini, 2024-02-07:
Hi @FaiqueAli, 

Find the start address in the .build file and pass it in the command. 
I used below command and able to boot the os
qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -device loader,file=QNX-IFS,addr=0x100000,cpu-num=0 -nographic -global xlnx,zynqmp-boot.cpu-num=1 -global xlnx,zynqmp-boot.use-pmufw=true

FaiqueAli, 2024-02-07:
@charantejaabbavathini 

Thank for the rep
for which build file are you talking about?
I only have QNX-IFS file at hand..

Faiq

charantejaabbavathini, 2024-02-08:
@FaiqueAli 
you should specify the start addr like this
qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -device loader,file=QNX-IFS,addr=0x100000,cpu-num=0 -nographic -global xlnx,zynqmp-boot.cpu-num=0

the build file is the part of the BSP you will have to download from qnx software center (BSP for zcu102)
Try the mentioned address.

Thanks!

FaiqueAli, 2024-02-09:
@charantejaabbavathini thank you for more information.
I am able to create the VM and mounted the diskimage, but now I can not execute mkdir inorder to create directory structure.
any idea about that ?

rmadhu90, 2025-05-12:
> [@FaiqueAli](https://github.com/FaiqueAli) you should specify the start addr like this qemu-system-aarch64 -M xlnx-zcu102 -m 8G -serial mon:stdio -device loader,file=QNX-IFS,addr=0x100000,cpu-num=0 [...]

@charantejaabbavathini 
Can you please give the command for -M arm-generic-fdt with dtb file ?
Thanks