Xilinx/qemu #80: Remote for `dtc` submodule points to deprecated git.qemu-project.org

0xjmux, 2023-08-01:
Following [this guide on the xilinx-wiki](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/822312999/Building+and+Running+QEMU+from+Source+Code), I'm trying to get a working qemu executable to simulate some custom hardware. 

The `git submodule update --init dtc` step fails, with error:
```error
Submodule 'dtc' (git://git.qemu-project.org/dtc.git) registered for path 'dtc'
Cloning into '/[path]/xilinx-qemu/dtc'...
fatal: unable to connect to git.qemu-project.org:
git.qemu-project.org[0: 20.218.253.93]: errno=Connection timed out

fatal: clone of 'git://git.qemu-project.org/dtc.git' into submodule path '/[path]/xilinx-qemu/dtc' failed
```

When you navigate to https://git.qemu-project.org, it redirects to their [Gitlab](https://gitlab.com/qemu-project/dtc). 

Since the repo was migrated, git isn't able to fetch the submodules; if that link (and any others still pointing to the old location) could be updated, it should fix the issue. 

Also, is that configure script still working? Or is there another way to get a working xilinx-qemu binary? Thanks.

figlesia-xilinx, 2023-09-15:
Hi @0xjmux ,

Can you please retry? (Above worked for me)

```
$ git clone https://github.com/Xilinx/qemu.git
Cloning into 'qemu'...
remote: Enumerating objects: 656044, done.
remote: Counting objects: 100% (62504/62504), done.
remote: Compressing objects: 100% (13272/13272), done.
remote: Total 656044 (delta 50040), reused 57846 (delta 49001), pack-reused 593540
Receiving objects: 100% (656044/656044), 278.25 MiB | 1.97 MiB/s, done.
Resolving deltas: 100% (538728/538728), done.
$ cd qemu/
$ git submodule update --init dtc
Submodule 'dtc' (https://gitlab.com/qemu-project/dtc.git) registered for path 'dtc'
Cloning into '/home/testuser/qemu-build-test/qemu/dtc'...
Submodule path 'dtc': checked out 'b6910bec11614980a21e46fbccc35934b671bd81'
$ git rev-parse HEAD
174fc3b9c65ab66132d3ddf1e05c0dfc14a8edfd
```
Best regards,
Francisco Iglesias

AbulesiKanJiJi, 2024-02-22:
Hi! Did you solve it? I meet the same problem. But retrying doesnt work.