Xilinx/qemu #87: QEMU  - How to enable two GEMs on xilinx-zcu102 machine

vladDmit, 2023-11-22:
By default only gem3 is enable on xilinx-zcu102 machine. How do I enable a second gem? so that my VM has two network interfaces : xzynq0 and xzynq1. Thanks! for your help.

saipava, 2023-11-23:
QEMU does emulate all the gems may be mdio connection might be missing. Have a look at here how gem3 is connected with phy
https://github.com/Xilinx/qemu-devicetrees/blob/master/board-zynqmp-zcu102.dts#L441

You need to also enable those in linux dts to access those in vm.

vladDmit, 2023-11-24:
Thanks for the response. I'm not using a linux image (it's qnx). The boot stall after cpu 1, when I use the machine type arm-generic-fdt. So I used the hard-coded machine xlnx-zcu102 which boots successfully. qemu ignore when I pass the option hw-dtb board-zynqmp-zcu102.dtb.  The machine xlnx-zcu102 is not dts based.. since dumpdtb return nothing. I'm still looking, but not  successful on which part of the code to update to get the second GEM functional for xlnx-zcu102

saipava, 2023-11-24:
For mainline version of zcu102 always have 4 gem's emulated.
https://github.com/Xilinx/qemu/blob/master/hw/arm/xlnx-zynqmp.c#L606

the phy address here seems constant, i'm not sure how it goes. Can you change it a different address and set the same expectations from guest.

vladDmit, 2023-12-18:
Thanks!! for your help saipava... this issue can be closed. I had to update the  QNX Driver for XZYNQ UltraScale+ MPSoC Ethernet GEM. This driver reads GEM interface information from the hwinfo section of the syspage. 
The hwinfo section is populated in the board-specific Startup code.
If there is no GEM interface specified in the hwinfo section, the driver will default to using GEM3.