Xilinx/qemu #97: fix support for glibc >= 2.41

sadguitarius, 2025-04-28:
I'm unable to build Petalinux because the `sched_attr` struct in linux-user/syscall.c conflicts with the one provided by glibc 2.41, which is the current version on  Arch Linux (I know Arch is not officially supported but the rest of the build is working fine up to this point). I did try manually editing the repo with `petalinux-devtool modify` but that doesn't work on qemu-xilinx-native due to some submodule problems. As far as I can tell, updating here would be the easiest way to fix it.

See [here](https://patchew.org/QEMU/20241011193140.1047648-1-raj.khem@gmail.com/) for further explanation and the fix, which has already been added to the QEMU main branch. Thank you!

sadguitarius, 2025-04-28:
sorry, the actual commit on main if [here](https://gitlab.com/qemu-project/qemu/-/commit/27a8d899c7a100fd5aa040a8b993bb257687c393)

sadguitarius, 2025-04-28:
Ugh ok I tried submitting a PR and then realized not only was mine automatically closed but that someone else had already tried to do the same thing. The automated message redirects to the QEMU gitlab where this has already been fixed. Is there no way to submit a PR for the Xilinx fork??

Dishoungh, 2025-08-10:
@sadguitarius I'm also trying to get this to work on PetaLinux/Yocto. How were you able to patch this? I'm on 2024.2/scarthgap.

Dishoungh, 2025-08-10:
Nevermind. As soon as I asked that question, I figured it out. For some reason, that happens a lot lol.

In case if someone else needs this, which I take it they will because at the time of writing this, Xilinx still hasn't patched this:

1. Go to the [QEMU commit link that fixes this](https://gitlab.com/qemu-project/qemu/-/commit/27a8d899c7a100fd5aa040a8b993bb257687c393)
2. Download the patch file (should be a small .patch file)

[image]

3. In your own layer meta-(WHATEVER), create recipes-devtools/qemu/patches
4. Copy patch file and name it whatever, I named it "0001-qemu-Do-not-define-struct-sched_attr.patch"
5. In meta-(WHATEVER)/recipes-devtools/qemu/qemu-xilinx-native_%.bbappend add these lines:

[image]

And that fixed it for me.

sadguitarius, 2025-08-10:
Glad you got it sorted out! Looks like this is still busted in 2025.1. FWIW I finally gave up and started using a virtual machine for Yocto/Petalinux running Ubuntu LTS because of little annoying things like this. Good luck!

Dishoungh, 2025-08-10:
At least the issue is relatively minor and that seems to be the only issue I came across on my host system (Gentoo) and my AlmaLinux VM.

I'm only messing around so I can try and convert my PetaLinux workflow to a native Yocto one. At least this will still be good information in case if I need to patch something later in the future with either PetaLinux or native Yocto.