Xilinx/qemu #104: Unable to clone u-boot submodule

rb1, 2025-09-22:
Hi,

I am not sure what's changed, but i'm unable to clone the repository and populate the submodules. The initial clone is fine, but initialising the submodules fails asking for authentication. 

`→ git submodule update --init --recursive
Submodule 'roms/QemuMacDrivers' (https://gitlab.com/qemu-project/QemuMacDrivers.git) registered for path 'roms/QemuMacDrivers'
Submodule 'roms/SLOF' (https://gitlab.com/qemu-project/SLOF.git) registered for path 'roms/SLOF'
Submodule 'roms/edk2' (https://gitlab.com/qemu-project/edk2.git) registered for path 'roms/edk2'
Submodule 'roms/ipxe' (https://gitlab.com/qemu-project/ipxe.git) registered for path 'roms/ipxe'
Submodule 'roms/openbios' (https://gitlab.com/qemu-project/openbios.git) registered for path 'roms/openbios'
Submodule 'roms/opensbi' (https://gitlab.com/qemu-project/opensbi.git) registered for path 'roms/opensbi'
Submodule 'roms/qboot' (https://gitlab.com/qemu-project/qboot.git) registered for path 'roms/qboot'
Submodule 'roms/qemu-palcode' (https://gitlab.com/qemu-project/qemu-palcode.git) registered for path 'roms/qemu-palcode'
Submodule 'roms/seabios' (https://gitlab.com/qemu-project/seabios.git/) registered for path 'roms/seabios'
Submodule 'roms/seabios-hppa' (https://gitlab.com/qemu-project/seabios-hppa.git) registered for path 'roms/seabios-hppa'
Submodule 'roms/skiboot' (https://gitlab.com/qemu-project/skiboot.git) registered for path 'roms/skiboot'
Submodule 'roms/u-boot' (https://gitlab.com/qemu-project/u-boot.git) registered for path 'roms/u-boot'
Submodule 'roms/u-boot-sam460ex' (https://gitlab.com/qemu-project/u-boot-sam460ex.git) registered for path 'roms/u-boot-sam460ex'
Submodule 'roms/vbootrom' (https://gitlab.com/qemu-project/vbootrom.git) registered for path 'roms/vbootrom'
Submodule 'tests/lcitool/libvirt-ci' (https://gitlab.com/libvirt/libvirt-ci.git) registered for path 'tests/lcitool/libvirt-ci'
Cloning into '/home/james/sources/qemu/roms/QemuMacDrivers'...
Cloning into '/home/james/sources/qemu/roms/SLOF'...
Cloning into '/home/james/sources/qemu/roms/edk2'...
Cloning into '/home/james/sources/qemu/roms/ipxe'...
Cloning into '/home/james/sources/qemu/roms/openbios'...
Cloning into '/home/james/sources/qemu/roms/opensbi'...
Cloning into '/home/james/sources/qemu/roms/qboot'...
Cloning into '/home/james/sources/qemu/roms/qemu-palcode'...
Cloning into '/home/james/sources/qemu/roms/seabios'...
Cloning into '/home/james/sources/qemu/roms/seabios-hppa'...
Cloning into '/home/james/sources/qemu/roms/skiboot'...
Cloning into '/home/james/sources/qemu/roms/u-boot'...
Username for 'https://gitlab.com':`

I used my browser to create an account and authenticate to gitlab. I then navigated to https://gitlab.com/qemu-project/u-boot.git only to get a 404 error.

I've observed this on the master branch by manually trying and also on the xlnx_rel_v2022.2 branch when trying to clone/fetch using yocto.

rb1, 2025-10-01:
uboot seems good now, but i still can't run the command above successfully because a different submodule is failing.

https://github.com/Zeex/subhook/ (qemu/roms/edk2/UnitTestFrameworkPkg/Library/SubhookLib/subhook)

I think this is a problem in edk2 repository.

kylex2, 2025-10-01:
Yes, ed2k now provides their own mirror of subhook at https://github.com/tianocore/edk2-subhook.git.

```
MIRRORS:append = " gitsm://github.com/Zeex/subhook.git gitsm://github.com/tianocore/edk2-subhook.git \
"
```